sandeepmistry/arduino-nRF5 #1: Blinky?

AdySan, 2016-03-29:
Just got this setup according to the ReadMe. Based on the following output seems to be building and downloading ok. For a simple Blink.ino example, what PIN should I use? 13 doesnt seem to work, not sure how these nrf52 pins are mapped.

```
GNU ARM Eclipse 64-bits Open On-Chip Debugger 0.10.0-dev-00287-g85cec24 (2016-01-10-12:03)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 0
swd
cortex_m reset_config sysresetreq
adapter speed: 1000 kHz
nrf51.cpu: target state: halted
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000574 msp: 0x20008000
** Programming Started **
auto erase enabled
wrote 4096 bytes from file /var/folders/75/wvhmv78s3zz48zf4fssk4__40000gn/T/buildd5ae82e677127ac286f51de66781d807.tmp/Blink.ino.bin in 0.232602s (17.197 KiB/s)
** Programming Finished **
** Verify Started **
verified 3388 bytes in 0.085533s (38.682 KiB/s)
** Verified OK **
** Resetting Target **
shutdown command invoked
```

AdySan, 2016-03-29:
These don't work btw.

```
// LEDs
#define PIN_LED1                (17)
#define PIN_LED2                (18)
#define PIN_LED3                (19)
#define PIN_LED4                (20)
#define LED_BUILTIN             PIN_LED1
```

AdySan, 2016-03-29:
FWIW, the example hex files located here from SDK 11.0 work fine, so hardware is ok.
```
/Users/adityatannu/Developer/nRF5_SDK_11.0.0_89a8197\ /examples/peripheral/blinky/hex 
```

sandeepmistry, 2016-03-30:
@AdySan check the back of the board, they are are actually connected to P0.17-20 which map to Arduino pin connectors D6-D9.

closed 2016-03-30 by sandeepmistry

AdySan, 2016-03-30:
Yes, will test soon 
[image]

AdySan, 2016-03-31:
How do I work on the latest core? Looks like I'm not getting the latest changes from the core by placing it in `hardware/nRF5` and updating it via git. I've also installed using Boards Manager. Do I need to rebuild the core somehow?

commit ec3c8eb5 (2016-03-30): Correct PIN_LEDn values

Resolves #1.