sandeepmistry/arduino-nRF5 #2: How to use latest core

AdySan, 2016-04-03:
Can you please give some details on how to work on the latest core from git?

I tried cloning this repo to 
`/Users/adityatannu/Documents/hardware/nRF5` 
and 
`/Users/adityatannu/Documents/Arduino/hardware/nRF5`

But I do not seem to be getting the latest changes in the IDE.

I'm not really sure how this works but for ESP8266/Arduino core, it is mapped to
`/Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266 `, I tried doing something similar but didn't work. The release I installed from Boards Manager seems to be the active one.

I'd really appreciate some guidance here!

bojh, 2016-04-05:
The users default SKETCHBOOK location (windows: C:\Users\<user-xy>\Documents\Arduino ) subdirectory `hardware\nRF5` has lower read prio than the board manager installation directory C:\Users\<user-xy>\AppData\Local\Arduino15\packages\nRF5... 
So I have saved the original version directory `0.0.0` as a zip file and cloned the arduino-nRF5  project in `0.0.0` dirctory place.
`C:\Users\<user-xy>\AppData\Local\Arduino15\packages\nRF5\hardware\nRF5\0.0.0`
With that setting you are directly able to pull new project changes.

AdySan, 2016-04-05:
@bojh aha, thanks a lot. I'm not sure where the corresponding location on OS X will be but I'll look it up. Thanks!

AdySan, 2016-04-06:
@bojh Cool, works on OS X too! I had to uninstall the one from Boards Manager and check out this repo here: 

`/Users/adityatannu/Documents/Arduino/hardware/nRF5/nRF5`

AdySan, 2016-04-06:
Not really sure where to copy the build tools though, it can't find the arm-g++ compiler anymore.

closed 2016-04-10 by AdySan

sandeepmistry, 2016-04-21:
I've made some clarifications to the read me in https://github.com/sandeepmistry/arduino-nRF5/commit/002f43a20e052d9718beb5bdfc55c0bd2cd57f98.

sandeepmistry, 2016-04-21:
I've updated the board manager with a 0.1.0 release. However, since I've made a breaking change to the provider name, I suggest you delete the 0.0.0 release and follow instructions from the read me from scratch.