sandeepmistry/arduino-nRF5 #3: Support for analogWrite with 3rd parameter frequency

bojh, 2016-04-05:
Motivated by a real application need (use case) and the particle (extended) [Wiring-API](https://docs.particle.io/reference/firmware/photon/#analogwrite-pwm-), an additional analogWrite(pin,pwm, freq) Wiring method would be very useful. One unique method with a default parameter set to 500Hz would also be valid.

bojh, 2016-04-12:
Some progress is done, but some mysterious artefacts are still there: 
1. I can't set to different pwm port, becaause one does modify the other, independend from the frequency!
Here is the test code: with an additional public method with an additional frequency parameter:

```
void analogWrite( uint32_t ulPin, uint32_t ulValue )
{
  analogWriteFreq(ulPin, ulValue, 500);
}

void analogWriteFreq( uint32_t ulPin, uint32_t ulValue, uint32_t ulFreq)
{
  if (ulPin >= PINS_COUNT) {
    return;
  }
  ulPin = g_ADigitalPinMap[ulPin];

  for (int i = 0; i < PWM_COUNT; i++) {
    if (pwmChannelPins[i] == NRF_PWM_PIN_NOT_CONNECTED || pwmChannelPins[i] == ulPin) {
      pwmChannelPins[i] = ulPin;      
      
	  NRF_PWM_Type* pwm = pwms[i];

      nrf_pwm_pins_set(pwm, pwmChannelPins);
      nrf_pwm_enable(pwm);

	  uint32_t ofrq = ulFreq;![Uploading nRF52-pwm-artefact.PNG…]()

	  nrf_pwm_clk_t baseClockEnum = NRF_PWM_CLK_16MHz;
	  if (ulFreq >512) {
		  uint32_t maxFreq = 16000000/((1<<writeResolution)-1);
		  if (ulFreq>maxFreq) ulFreq=maxFreq;	//writeRes 8:62745Hz; 9:31331Hz 10:15640Hz
		  baseClockEnum = NRF_PWM_CLK_16MHz;                   //DIV2  =0
	  }
	  else if (ulFreq >256) baseClockEnum = NRF_PWM_CLK_8MHz;  //DIV4  =1
	  else if (ulFreq >128) baseClockEnum = NRF_PWM_CLK_4MHz;  //DIV8  =2
	  else if (ulFreq > 64) baseClockEnum = NRF_PWM_CLK_2MHz;  //DIV16 =3
	  else if (ulFreq > 32) baseClockEnum = NRF_PWM_CLK_1MHz;  //DIV32 =4
	  else if (ulFreq > 16) baseClockEnum = NRF_PWM_CLK_500kHz;//DIV64 =5
	  else if (ulFreq >  8) baseClockEnum = NRF_PWM_CLK_250kHz;//DIV128=6
	  else {
		if (ulFreq < 4) ulFreq = 4;				//
	    baseClockEnum = NRF_PWM_CLK_125kHz;                    //DIV256=7
	  }
	  uint32_t topCntValue = 16000000/(ulFreq << baseClockEnum); 	// max. 15bit! = 32767!
      nrf_pwm_configure(pwm, baseClockEnum, NRF_PWM_MODE_UP, (uint16_t)topCntValue);

	  uint32_t v = ((topCntValue)*ulValue) / ((1<<writeResolution)-1);
	  if (v > topCntValue) {
	  	  v = topCntValue;
	  }
      pwmChannelSequence[i] = (uint16_t)v;	// max 15bit!
	  pwmChannelSequence[i]|= 1<<15;	    // Arduino ON/OFF duty cycle (set invert bit)

      nrf_pwm_loop_set(pwm, 0);
      nrf_pwm_decoder_set(pwm, NRF_PWM_LOAD_COMMON, NRF_PWM_STEP_AUTO);
	  nrf_pwm_seq_ptr_set(pwm, 0, &pwmChannelSequence[i]);
      nrf_pwm_seq_cnt_set(pwm, 0, 1);
      nrf_pwm_seq_refresh_set(pwm, 0, 1);
      nrf_pwm_seq_end_delay_set(pwm, 0, 0);
      nrf_pwm_task_trigger(pwm, NRF_PWM_TASK_SEQSTART0);

      break;
    }
  }
}

```

Trying that with the following example:

```
void setup() {
  pinMode( 9, OUTPUT);
  pinMode(10, OUTPUT);
  pinMode(11, OUTPUT);
  analogWriteFreq(11, 170, 2000);
  analogWriteFreq(10,  99, 500);
  analogWriteFreq( 9,  37, 50);
  delay(2000);
  analogWriteFreq(11,  30, 2000);
}

```
.... does lead to an curious artefact: see the traced output!

[image]

bojh, 2016-04-12:
unintentially closed

bojh, 2016-04-12:
[image]

sandeepmistry, 2016-04-21:
@bojh good suggestion, but I would prefer to keep the existing ```analogWrite``` API the same.

I think it would better to have a global ```analogFrequency``` API, this also makes things easier on the nRF51 side. This lines up with the [analogReference](https://www.arduino.cc/en/Reference/AnalogReference) and [analogWriteResolution](https://www.arduino.cc/en/Reference/AnalogWriteResolution) API's.

bojh, 2016-04-21:
Hi Sandeep,
I currently awaiting my company open source contribution approval, that I would be allowed to contribute to your project.
I have done a complete, fully working pwm API. As soon as I can commit I can show you complete but here the ideas as follows.

1.      Keep ADUINO compatibility!

2.      Add optional additional functionality and usage freedom

3.      Use Nordic PWM options

Proposed API looks currently as follows:

//standard Arduino API:
void analogWrite(uint32_t ulPin, uint32_t ulValue): // standard  ARDUINO
//…will 1st time call initPWm() with 500Hz and internally analogWritePwm().

// new additional functions:
void analogWritePwm(uint32_t ulPin, uint32_t ulValue, uint8_t pwmId);
void initPwm  (uint8_t pwmId, uint32_t ulFreq);
void deinitPwm(uint8_t pwmId);

This allows usage of all 3 PWM’s with indiv. frequency and upto 4 Pin channels with individual duty-cycle values for each PWM.
Good ideas are welcome - naming and parameter sequence can also be discussed.

Hope, I will get my OS approval soon.

Cheers.

-

Mit freundlichen Grüßen / Best regards

Jörg Hubschneider

Bosch Software Innovations GmbH
Edge Device Integration (INST/ESY1)
Stuttgarter Straße 130
71332 Waiblingen/GERMANY
www.bosch-si.com<http://www.innovations.de/>
www.blog.bosch-si.com<http://www.blog.bosch-si.com/>

Tel. +49 (711) 811 58104
Fax. +49 (711) 811 58200
joerg.hubschneider@bosch-si.com<mailto:joerg.hubschneider@bosch-si.com>

Registered office: Berlin, Register court: Amtsgericht Charlottenburg; HBR 148411 B;
Executives: Dr.-Ing. Rainer Kallenbach, Michael Hahn

sandeepmistry, 2016-04-25:
@bojh cool, please keep the nRF51 in mind: https://github.com/sandeepmistry/arduino-nRF5/blob/master/cores/nRF5/wiring_analog_nRF51.c

It currently uses a single timer, so individual frequency per pin complicates things.

sandeepmistry, 2016-12-18:
I'm closing this for now, the official Arduino SAMD core does not provide this feature, and based on my comment above, it would be difficult to implement on nRF51 boards too.

closed 2016-12-18 by sandeepmistry

bojh, 2016-12-19:
@sandeepmistry by the way: I'm working with a local branch supporting the posibility of frequency configuration - meanwhile with a float frequency variable for fine lower frequency ajustment. I read the library vs. extension discussion. My initial recommendation was to extent the core API, but if we want to follow with a strict common core API, we can think about an optional PWM library. My current solution works as core extension, enabling a fully flexible nRF5 NRF_PWM_Type (NRF_PWM0..2) management for frequncies between (4Hz..62kHz), where the ordenary analogWrite() method does use as default PWM0 with a default frequency of 500Hz. If we define, PWM0 is fix reserved for the core standard API, we are still able to prepare the core for an addional library, optionally taking only PWM1 and PWM2 for flexible frequency setting. What is your opinion? For a better imagination, I have attached an alternative wiring_analog_nRF52.c file. For to test, you are free to #define NEW_PWM and wiring-analog.h modification. Any suggestions for an additional PWM library API, is there a common used API?
[wiring_analog_nRF52.zip](https://github.com/sandeepmistry/arduino-nRF5/files/660646/wiring_analog_nRF52.zip)