sandeepmistry/arduino-nRF5 #5: Using Wire

AdySan, 2016-04-10:
`variants.h` seems to have these pin definitions for SDA, SCL. What pins should be used on the nRF52 DK, (26) and (27)?

```
#define WIRE_INTERFACES_COUNT 1

#define PIN_WIRE_SDA         (20u)
#define PIN_WIRE_SCL         (21u)
```

`Wire.cpp` has this way to instantiate a Wire interface but doesn't seem to be called anywhere, so I'm not sure what values `_uc_pinSCL` and `_uc_pinSDA` take in the `TwoWire::begin` call

```
TwoWire::TwoWire(NRF_TWIM_Type * p_twim, NRF_TWIS_Type * p_twis, IRQn_Type IRQn, uint8_t pinSDA, uint8_t pinSCL)
```

bojh, 2016-04-20:
Hi, I'm also currently testing I2C implementation
1. I can confirm the pin setting is correct. On Arduino pin 20=SDA an 21=SCL. This relates to nordic chip pins P.26=SCL and P.27=SDA => OK

2. having setup the compile issue from your PR ...
...write registers seems working.
3. .. but: Reading a register does allways end up in non available bytes and reading than does result in a FF byte value. So I'm just in there to debug what happens or what is missing.

bojh, 2016-04-20:
I saw in the Nordic SDK documentation, that TWIM0/TWMS0 does share the same resources than SPIM0/SPIS0, wich is currently implemented as the standard SPI Arduino Interface. I would propose the using of disjunct resources for standard Wire and SPI interface by moving  TWI resource to TWMM1/TWIS0. In the Wire.cpp implementation, following has to be changed to:

```
TwoWire Wire  (NRF_TWIM1, NRF_TWIS1, SPIM1_SPIS1_TWIM1_TWIS1_SPI1_TWI1_IRQn, PIN_WIRE_SDA, PIN_WIRE_SCL);

extern "C"
{
  void SPIM1_SPIS1_TWIM1_TWIS1_SPI1_TWI1_IRQHandler(void)
  {
    Wire.onService();
  }
}

```

sandeepmistry, 2016-04-21:
@AdySan please see existing Wire examples: Examples -> Wire

```Wire``` instance is created automatically: https://github.com/sandeepmistry/arduino-nRF5/commit/f476e0ce355fbc223f354882a2e9ec074ee74a92

@bojh good suggestion! I've made the change in https://github.com/sandeepmistry/arduino-nRF5/commit/ff441d3d06660f189024a6d3875496908b232b60

Please open a new issue to track your observations from https://github.com/sandeepmistry/arduino-nRF5/issues/5#issuecomment-212399441.

closed 2016-04-21 by sandeepmistry