sandeepmistry/arduino-nRF5 #7: Wire nRF52 DK implementation bug

bojh, 2016-04-21:
I expect I have found the nRF52 DK Wire issue: SCL & SDA are swapped! 
SCL is on the the SDA hardware pin - as I have confirmed with my Salea Logic !
In `variants.cpp` the Arduino pin translation should be:

```
  // SDA, SCL
  26,	//SDA
  27,	//SCL

```
... but I'm currently not able to test and confirm it, because I have a (local?) build flushing issue with the new board management under (sandeepmistry/nRF5): my new *.ino.hex images are not flushed correctly!

bojh, 2016-04-21:
Since I can flush my nRF52-DK again:  The above mentioned SDA<->SCL pin swap is definitfly a bug.
But the next bug is: The Wire.endTransmission(stopbit) function does loop at:

```
  while(!nrf_twim_event_check(_p_twim, NRF_TWIM_EVENT_LASTTX) && !nrf_twim_event_check(_p_twim, NRF_TWIM_EVENT_ERROR));
  nrf_twim_event_clear(_p_twim, NRF_TWIM_EVENT_LASTTX);
```
... sometimes! (after it have passed this location 2 times!

bojh, 2016-04-22:
Here are the I2C logs from the logic:
nominal (RBL nRF52822):
```
1.340522583333333, 0,p,'128',Write,ACK
1.340622583333333, 0,p,    (,Write,ACK
1.343231250000000, 1,p,'168',Write,ACK
1.345838916666667, 2,q, '17',Read, NAK
1.354634750000000, 3,p,'163',Write,ACK
1.357241916666667, 4,q,  '6',Read, NAK
2.701635041666667, 5,p,  '2',Write,ACK
2.704242708333333, 6,q,'255',Read, NAK
2.712511791666667, 7,p,  '2',Write,ACK
```
and nRF52 implementation:
```
1.186645708333333,1,p,'128',Write,ACK
1.186746083333333,1,p,(,Write,ACK
1.191903333333333,3,p,'168',Write,ACK
1.197027666666667,4,q,'17',Read,NAK

```
and at
```
    uint8_t x = Wire.read();
    Serial.print("$"); Serial.print(reg, HEX); Serial.print(": 0x"); Serial.println(x, HEX);
    err = Wire.endTransmission();

```
the want read byte is  transfered - but endTransmission does loop!

bojh, 2016-04-22:
commenting out following while loop line in the endTransmission(stopBit) implementation
... does work for this example.

```
..
  while(!nrf_twim_event_check(_p_twim, NRF_TWIM_EVENT_TXSTARTED) && !nrf_twim_event_check(_p_twim, NRF_TWIM_EVENT_ERROR));
  nrf_twim_event_clear(_p_twim, NRF_TWIM_EVENT_TXSTARTED);
//>>  while(!nrf_twim_event_check(_p_twim, NRF_TWIM_EVENT_LASTTX) && !nrf_twim_event_check(_p_twim, NRF_TWIM_EVENT_ERROR)); //<<
  nrf_twim_event_clear(_p_twim, NRF_TWIM_EVENT_LASTTX);

  if (stopBit || nrf_twim_event_check(_p_twim, NRF_TWIM_EVENT_ERROR))
..

```
The question is: Where to find what is the nominal right way?

bojh, 2016-04-22:
may be there is a problem:

[see nordic devzone](https://devzone.nordicsemi.com/question/68741/twi-issue-migrating-sdk8-sdk11/)

but now?

prjh, 2016-04-22:
It seems it is an timing issue:
Folloing does work with 100kHz upto 400kHz: A small delay behind `nrf_twim_tx_buffer_set()` is the real location 
... and does help!
Can we also wait for an event?

```
// Errors:
//  0 : Success
//  1 : Data too long
//  2 : NACK on transmit of address
//  3 : NACK on transmit of data
//  4 : Other error
uint8_t TwoWire::endTransmission(bool stopBit)
{
  transmissionBegun = false ;

  // Start I2C transmission
  nrf_twim_address_set(_p_twim, txAddress);

  nrf_twim_task_trigger(_p_twim, NRF_TWIM_TASK_RESUME);
  nrf_twim_task_trigger(_p_twim, NRF_TWIM_TASK_STARTTX);

  nrf_twim_tx_buffer_set(_p_twim, txBuffer._aucBuffer, txBuffer.available());
  delay(1);   //<<<<<
  //delayMicroseconds(500); // not supported!

  nrf_twim_task_trigger(_p_twim, NRF_TWIM_TASK_STARTTX);

  while(!nrf_twim_event_check(_p_twim, NRF_TWIM_EVENT_TXSTARTED) && !nrf_twim_event_check(_p_twim, NRF_TWIM_EVENT_ERROR));
  nrf_twim_event_clear(_p_twim, NRF_TWIM_EVENT_TXSTARTED);
  while(!nrf_twim_event_check(_p_twim, NRF_TWIM_EVENT_LASTTX) && !nrf_twim_event_check(_p_twim, NRF_TWIM_EVENT_ERROR));
  nrf_twim_event_clear(_p_twim, NRF_TWIM_EVENT_LASTTX);

  if (stopBit || nrf_twim_event_check(_p_twim, NRF_TWIM_EVENT_ERROR))
  {
    nrf_twim_task_trigger(_p_twim, NRF_TWIM_TASK_STOP);
    while(!nrf_twim_event_check(_p_twim, NRF_TWIM_EVENT_STOPPED));
    nrf_twim_event_clear(_p_twim, NRF_TWIM_EVENT_STOPPED);
  }
  else
  {
    nrf_twim_task_trigger(_p_twim, NRF_TWIM_TASK_SUSPEND);
    while(!nrf_twim_event_check(_p_twim, NRF_TWIM_EVENT_SUSPENDED));
    nrf_twim_event_clear(_p_twim, NRF_TWIM_EVENT_SUSPENDED);
  }

  if (nrf_twim_event_check(_p_twim, NRF_TWIM_EVENT_ERROR))
  {
    nrf_twim_event_clear(_p_twim, NRF_TWIM_EVENT_ERROR);

    uint32_t error = nrf_twim_errorsrc_get_and_clear(_p_twim);

    if (error == NRF_TWIM_ERROR_ADDRESS_NACK)
    {
      return 2;
    }
    else if (error == NRF_TWIM_ERROR_DATA_NACK)
    {
      return 3;
    }
    else
    {
      return 4;
    }
  }

  return 0;
}

```

sandeepmistry, 2016-04-25:
@bojh I've correct the pins in https://github.com/sandeepmistry/arduino-nRF5/commit/888798d6a42962c15cff91e78cb68f48b19b87e9

sandeepmistry, 2016-04-25:
@prjh that is very cool!

> Can we also wait for an event?

Not that I'm aware of.

I'll try this again with the bundled Wire examples later this week.

sandeepmistry, 2016-04-26:
I wonder if it's the double STARTTX:

```c
  nrf_twim_task_trigger(_p_twim, NRF_TWIM_TASK_RESUME);
  nrf_twim_task_trigger(_p_twim, NRF_TWIM_TASK_STARTTX);

  nrf_twim_tx_buffer_set(_p_twim, txBuffer._aucBuffer, txBuffer.available());

  nrf_twim_task_trigger(_p_twim, NRF_TWIM_TASK_STARTTX);
```

maybe it should be:

```c
  nrf_twim_task_trigger(_p_twim, NRF_TWIM_TASK_RESUME);

  nrf_twim_tx_buffer_set(_p_twim, txBuffer._aucBuffer, txBuffer.available());

  nrf_twim_task_trigger(_p_twim, NRF_TWIM_TASK_STARTTX);
```

bojh, 2016-04-26:
@sandeepmistry good idea -seems valuable
but I didn't get it working by commenting out one of the STARTX triggers - nor to avoid the delay.
The only way at the moment is as follows:

```
  nrf_twim_task_trigger(_p_twim, NRF_TWIM_TASK_RESUME);
  nrf_twim_task_trigger(_p_twim, NRF_TWIM_TASK_STARTTX);

  nrf_twim_tx_buffer_set(_p_twim, txBuffer._aucBuffer, txBuffer.available());
  uint32_t us=micros();while ((micros()-us)<125);  //<<<<< 100kHz: 125us; 400kHz: 32us

  nrf_twim_task_trigger(_p_twim, NRF_TWIM_TASK_STARTTX);

```

bojh, 2016-05-03:
Starting with the commit eb01946 (removed HAL-API) the `Wire::endTransmission()` method does endup in with return code=2 (NRF_TWIM_ERROR_ADDRESS_NACK) and can' be fixed with the delay work around.

sandeepmistry, 2016-05-07:
@bojh could you please clarify your comment in https://github.com/sandeepmistry/arduino-nRF5/issues/7#issuecomment-216461007. Maybe share the sketch you are using?

For me both the master receiver and master sender examples are working, I have it connected to Arduino Zero running the slave sender and slave receiver.

@prjh would you be able to try out the latest master version?

bojh, 2016-05-09:
@sandeepmistry My test hardware is the [Adafruit Cap.-Touch](https://www.adafruit.com/product/1947) Display with the driver libs Adafruit_ILI9341 and Adafruit_FT6206 last version. ILI9341 needs smaller modifications for HW SPI and FT9206 can be additionally trimmed to I2C 400kHz. I got it working again (100kHz & 400kHz) on the current commit 233eb17 base with the following `Wire_nrf52.cpp` modification for the `Wire::endTransmission()` method:

```
  while(!_p_twim->EVENTS_TXSTARTED && !_p_twim->EVENTS_ERROR);
  _p_twim->EVENTS_TXSTARTED = 0x0UL;
  
  uint32_t t = micros(); while((micros()-t)<40);	//bojh
  //bojh while(!_p_twim->EVENTS_LASTTX && !_p_twim->EVENTS_ERROR);
  _p_twim->EVENTS_LASTTX = 0x0UL;

  if (stopBit || _p_twim->EVENTS_ERROR)

```
It seems that there is a special timing issue: 
1. The while loop break condition  `while(!_p_twim->EVENTS_LASTTX .&& ...)` is newer reached at that Wire.read() endTransmission() command - with or without the delay.
2. But a small delay is necessary - for a 100kHz & 400kHz clock!
... with that real touch-hardware. 
My modified FT6206::begin() method is as follows:
```
boolean Adafruit_FT6206::begin(uint8_t threshhold) {
#if defined (NRF51)  // RBL
    Wire.begin(TWI_SCL, TWI_SDA, TWI_FREQUENCY_FREQUENCY_K400);
#elif defined(NRF52)
  Wire.begin();
  Wire.setClock(400000);
#else
  Wire.begin();
#endif

  // change threshhold to be higher/lower
  writeRegister8(FT6206_REG_THRESHHOLD, threshhold);
  
  if ((readRegister8(FT6206_REG_VENDID) != 17) || (readRegister8(FT6206_REG_CHIPID) != 6)) return false;

```
while:

```
uint8_t Adafruit_FT6206::readRegister8(uint8_t reg) {
  uint8_t x ;
   // use i2c
    Wire.beginTransmission(FT6206_ADDR);
    Wire.write((byte)reg);
    Wire.endTransmission();
    Wire.beginTransmission(FT6206_ADDR);
    Wire.requestFrom((byte)FT6206_ADDR, (byte)1);
    x = Wire.read();
    Wire.endTransmission();

  //  Serial.print("$"); Serial.print(reg, HEX); 
  //  Serial.print(": 0x"); Serial.println(x, HEX);
  
  return x;
}

```

sandeepmistry, 2016-05-15:
@bojh from my understanding, this is not proper use of the Wire API:

```cpp
    Wire.beginTransmission(FT6206_ADDR);
    Wire.write((byte)reg);
    Wire.endTransmission();

    Wire.beginTransmission(FT6206_ADDR);
    Wire.requestFrom((byte)FT6206_ADDR, (byte)1);
    x = Wire.read();
    Wire.endTransmission();
```

```Wire.beginTransmission()``` and ```Wire.endTransmission();``` should only be used with ``write``, so you could try:

```cpp
    Wire.beginTransmission(FT6206_ADDR);
    Wire.write((byte)reg);
    Wire.endTransmission();

    Wire.requestFrom((byte)FT6206_ADDR, (byte)1);
    x = Wire.read();
```

We can try to make the way you were trying to work as well ... maybe only wait for last TX if there are bytes in the buffer to send. PR's welcome :)

bojh, 2016-05-17:
@sandeepmistry ... after my initial research, I can confirm that. - OK I took the adafruit code as is - but I will test the Wire.read() without the begin/endTransmission.

bojh, 2016-05-18:
@sandeepmistry Thanks for your help. I have modified the Adafruit FT6206 library to a Wire API conform `Wire.read()` handling without any starting beginTransmission() and endTransmission() around Wire.read(). The current commit dfe5fac does properly work out of the box as awaited :-).

closed 2016-05-18 by bojh

KivraS, 2022-02-25:
Hi. I am trying to use the Arduino_LSM6DSOX library on my Nrf52810 and i am facing the exact same issue @bojh menitioned .
When calling endTransmission it seems to be stuck on 
`while(!_p_twim->EVENTS_LASTTX && !_p_twim->EVENTS_ERROR);` at line 233