sandeepmistry/arduino-nRF5 #8: Flushing programs does not really happen not with new sandeepmistry/nRF5 0.1.0 board installation

bojh, 2016-04-21:
Under "Nordic Semiconductor nRF5 Boards": nrf52 DK (Version 0.1.0)
ii is possible to compile and do flushing the current program - but running is every time the previous old last flushed image!
I have done a complete de- and re-installation - still the same behaviour. Something is wrong in that new inspiring environment.

bojh, 2016-04-21:
Correct reading of the README does help :-) ! 

I have flushed the S132 Softdevice for nRF52-DK and...
... IT DOES WORK! 
Good work what you have done!

closed 2016-04-21 by bojh

sandeepmistry, 2016-04-25:
@bojh 

> Correct reading of the README does help :-) !

Cool, any suggestions on how the documentation or UX could be improved?

bojh, 2016-04-27:
One Remark: 
In the new approach with a installed local developer core image in the `SKETCHBOOK/hardware/sandeepmistry/nRF5` folder, I can find two identical `Nordic Semiconductor nRF5 Boards` groups in the `Board Manager`. The second one is the local developer ones.
To flush a SoftDevice via `nRF Flush SoftDevice`, it is necessary to select the first group board (e.g. `nRF52 DK`) instance in the `Board Manager`. Otherwise a error message wiil give a hint:

`nRF5 Flash SoftDevice is only supported on 'Nordic Semiconductor nRF5 Boards' boards!`

sandeepmistry, 2016-05-02:
See discussion in https://github.com/sandeepmistry/arduino-nRF5/issues/10#issuecomment-216083445 about the last comment.