sandeepmistry/arduino-nRF5 #10: Warnings for library classes: Wire and SPI

bojh, 2016-04-27:
Compiling an sketch does create following warnings:

```
WARNING: library SPI claims to run on [nRF5] architecture(s) and may be incompatible with your current board which runs on [nRF] architecture(s).
WARNING: library Wire claims to run on [nRF5] architecture(s) and may be incompatible with your current board which runs on [nRF] architecture(s).
```
There must be a mismatch in between a architcture naming `nRF` and the `nRF5`.

sandeepmistry, 2016-05-02:
@bojh could you please share all of the build output?

Maybe you core is in a ``nRF`` folder instead of ``nRF5``?

bojh, 2016-05-02:
You a right - thanks at all. My folder is the `nRF` one - and I know than why I got 2 Nordic board groups in the board selection of the manager menu :-) But this is not so bad for to switch over between the developer and manager core version.

closed 2016-05-02 by bojh