sandeepmistry/arduino-nRF5 #11: nRF52 analogRead() does not work properly

bojh, 2016-04-29:
Reading analog values on analog ports does lead to completly cracy values and it seems that reading values from different port does influence one value the other.

bojh, 2016-05-02:
I have tested the new commit (eb01946) with removed "driver_nrf//hal". The issue is still there. 
What are the reasons for to remove the hal layer? .... too small advantage?

sandeepmistry, 2016-05-07:
@bojh I think https://github.com/sandeepmistry/arduino-nRF5/commit/233eb177105157f9e6d687550baf8ca9869ed9ec makes things better, please try.

> What are the reasons for to remove the hal layer?

Licensing.

bojh, 2016-05-09:
... not really! - In my application I have a 100k poti for 0...3,3V to one of the analog ports and I got only in the lower range ~ 0..0.643V => 0-1023 digits! a proportional value. (based on commit 233eb17)

prjh, 2016-05-09:
I have looked up into the nordic API docu. The diver [API ](http://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk5.v11.0.0%2Findex.html)does initialize the GAIN with GAIN1_6 (1/6). Seems reasonable. I will test that setting.

bojh, 2016-05-10:
@sandeep Proposal: add  a new global saadcGain variable, so we can ajust the gain with in combination within the `analogReference()` method e.g.:

```
/*
 * Internal Reference is at 0.6v!
 * External Reference should be between 1v and VDDANA-0.6v=2.7v
 *
 * Warning : On Arduino Zero board the input/output voltage for SAMD21G18 is 3.3 volts maximum
 */
void analogReference( eAnalogReference ulMode )
{
  switch ( ulMode ) {
    case AR_DEFAULT:
    case AR_INTERNAL:
    default:
      saadcReference = SAADC_CH_CONFIG_REFSEL_Internal;
      saadcGain           = SAADC_CH_CONFIG_GAIN_Gain1_5;	//=> 0...3,0V
      break;

    case AR_VDD4:
      saadcReference = SAADC_CH_CONFIG_REFSEL_VDD1_4;	
      saadcGain          = SAADC_CH_CONFIG_GAIN_Gain1_4;    //=> 0...VDD(3.3V)
      break;
  }
}
```
and:

```
uint32_t analogRead( uint32_t ulPin )
{
  ...
  NRF_SAADC->CH[0].CONFIG =   ((SAADC_CH_CONFIG_RESP_Bypass   << SAADC_CH_CONFIG_RESP_Pos)   & SAADC_CH_CONFIG_RESP_Msk)
                            | ((SAADC_CH_CONFIG_RESP_Bypass   << SAADC_CH_CONFIG_RESN_Pos)   & SAADC_CH_CONFIG_RESN_Msk)
                            | ((saadcGain                     << SAADC_CH_CONFIG_GAIN_Pos)   & SAADC_CH_CONFIG_GAIN_Msk)
                            | ((saadcReference                << SAADC_CH_CONFIG_REFSEL_Pos) & SAADC_CH_CONFIG_REFSEL_Msk)
                            | ((SAADC_CH_CONFIG_TACQ_3us      << SAADC_CH_CONFIG_TACQ_Pos)   & SAADC_CH_CONFIG_TACQ_Msk)
                            | ((SAADC_CH_CONFIG_MODE_SE       << SAADC_CH_CONFIG_MODE_Pos)   & SAADC_CH_CONFIG_MODE_Msk);
  NRF_SAADC->CH[0].PSELN = pin;
  NRF_SAADC->CH[0].PSELP = pin;
...

```

sandeepmistry, 2016-05-15:
Ok, I'll try with a 10k trimpot when I get a chance. Those comments in the code above also need to be updated, they are left overs from the SAMD core.

sandeepmistry, 2016-12-05:
@bojh can you submit a PR for this?

bojh, 2016-12-05:
Yes, I can do that :-)

sandeepmistry, 2016-12-08:
Fixed via #66.

closed 2016-12-08 by sandeepmistry