sandeepmistry/arduino-nRF5 #14: Question: How to flush RBL nRF51822 and BLENano?

bojh, 2016-05-12:
The arduino-nRF5 project can flush the nRF52-DK with it's buildin JLink debugger. Do I have to connect a external JLink to the RBL nRF51822 board by soldering a header to SWD-B? Or is the idea to support a serial programmer later on - especilly for BLENano?

sandeepmistry, 2016-05-16:
>  Do I have to connect a external JLink to the RBL nRF51822 board by soldering a header to SWD-B?

Nope, that board has a built-in CMSIS-DAP programmer on the USB port, just select it as the programmer from Tools -> Programmer to flash the soft device etc.

> Or is the idea to support a serial programmer later on - especilly for BLENano?

No plans for serial programmer, this would require use of a boot loader on the device.

The USB programmer board that comes with the Nano (MK20 based), is also a CMSIS-DAP programmer, so the above applies to it as well.

closed 2016-05-16 by sandeepmistry

prjh, 2016-05-16:
@sandeepmistry Ah OK Thanks at all for that help - I have  got a better view now. 
I have done that, but I'm still struggling against it - a simple "blink" or "Serial" example does not work - and I think the setting and initialization is all done well.