sandeepmistry/arduino-nRF5 #16: nRF51 and nRF52 Development Boards

kriswiner, 2016-05-26:
Hi Sandeep,

I have designed a couple of development boards using the nRF51 and nRF52; they have been for sale on [Tindie](https://www.tindie.com/products/onehorse/nrf52832-development-board/). Can I use your Arduino core to program them through the redBear programmer? Do I need to go through J-link on an nRF52 DK board? If I send you one of these boards would you help me learn how to program it using your core?

I am working with another fellow now to build an Arduino core for the STM32L4 family. This is the development board we are using for that:

https://www.tindie.com/products/onehorse/dragonfly-stm32l4-breakout-board/

Please let me know if you have some interest in collaborating to make the nRF52 usable with the Arduino IDE.

Kris

sandeepmistry, 2016-05-31:
> Can I use your Arduino core to program them through the redBear programmer? Do I need to go through J-link on an nRF52 DK board?

Yes, the BLE Nano's MK20 USB board is actually a CMSIS-DAP programmer that is compatible with openocd. Just connect VDD, GND, DIO, and CLK to your board, then select "CMSIS-DAP" in the IDE from Tools -> Programmer.

[image]

You can also use the JLink on the nRF52 DK, but will need to short some headers to enable external programming. If you go this route select "JLink" in the IDE from Tools -> Programmer.

> If I send you one of these boards would you help me learn how to program it using your core?

I'll propose the following trade: I'll help you to get your board added/working with this core, in exchange for you creating a guide on how to do so, so it's easier for others to add new boards in the future. Then you can send me a board if you like.

What do you think?

Does your board have a one to one mapping for broken out nRF52 pins? If so, I suggest you create a new variant based on the "Generic nRF51822" and "nRF52 DK" variants:
 * https://github.com/sandeepmistry/arduino-nRF5/blob/master/boards.txt#L180-L218
 * https://github.com/sandeepmistry/arduino-nRF5/blob/master/boards.txt#L22-L57

I think we can re-use the `variants/Generic/` folder, just need the nRF52 stuff from the DK entry in boards.txt.

sandeepmistry, 2016-06-20:
@kriswiner ping ...

kriswiner, 2016-06-23:
Hi Sandeep,

Thomas Roell and I are working to design an nRF52 Development Board that uses the STM32F042 as a boot loader and allows flashing using the Arduino IDE over the USB. Thomas plans to start with your Arduino core once we get the bootloader working and build from there. I think he has been in contact with you.

Kris

sandeepmistry, 2016-06-27:
Hi @kriswiner, thanks for the update, yes I have exchanged a few emails with Thomas in the past.

It looks like your initial question has been answered, so I'm going to close this issue.

closed 2016-06-27 by sandeepmistry