sandeepmistry/arduino-nRF5 #17: Compile error. Missing utoa

rogerclarkmelbourne, 2016-07-02:
If I clone the repo into the Arduino hardware folder, and try to compile a blank sketch with any of the boards I get an error about utoa not being available

It looks like its used here

https://github.com/sandeepmistry/arduino-nRF5/search?utf8=%E2%9C%93&q=utoa

So I tried installing from the JSON package and it worked OK

Note. I'm using IDE 1.6.9 as this generally works with all projects that work with 1.6.5

sandeepmistry, 2016-07-04:
Hi @rogerclarkmelbourne,

You'll need to get the toolchain from the JSON first prior to cloning from Github as per the read me. Newer versions of newlib include utoa now. Let me know if this can be clarified in the read me.

closed 2016-07-04 by sandeepmistry

rogerclarkmelbourne, 2016-07-04:
OK

I didn't try going back to the git clone after I'd installed the package from JSON

I'll try removing the core from the installed package, so I retain the tools and then try using the git clone again

sandeepmistry, 2016-07-13:
Hi @rogerclarkmelbourne, did this work for you?

rogerclarkmelbourne, 2016-07-13:
Yes. I did work for me in the end, by installing via JSON and then cloning the github repo to get the Generic board.

However it only works without the Softdevice, so is not much use to me at the moment.

As I need the SD and all the BLE API/SDK stuff, I've switched back to using the RBL nRF51822 Arduino repo, as it has recently been updated with up to date mbed internals

I'm going to update my fork of the RBL repo, as I've now done a load of work to speed up the compile by building a separate static lib for the mbed and BLE api code (which are 105 c and cpp files in their own right and take quite a long time to compile)

But I will keep an eye on how your repo develops.

Thanks

Roger