sandeepmistry/arduino-nRF5 #18: Operation on boards without 32kHz crystal

rogerclarkmelbourne, 2016-07-02:
Can you advise on getting this to work on generic boards which don't have the 32khz crystal?

nihaopaul, 2016-07-03:
I flashed my own designed  nrf52 without an external 32khz crystal and could use things like pwm after I flashed the soft device. Was there a specific function you are having issues with?

rogerclarkmelbourne, 2016-07-03:
It was a general question, as I couldnt get the core to work, but I've since tried to get it working on a board that has the 32khz crystal, but still no luck.

I'll need to do some more investigation.

BTW. I'm using a WaveShare board (nrf51822 QFAC), upload manually via JLink

I've tried the S130 soft device (for nrf51822) and also building for "none" on the soft device selector (as I realised the S132 soft device is not for the nrf51822)

So I will need to do some more experimentation to see if I can figure out why even a blink sketch isnt working

sandeepmistry, 2016-07-04:
Hi @rogerclarkmelbourne,

I've briefly tested with a [OSHChip](http://oshchip.org) which doesn't have a 32kHz crystal, no issues when none is selected.

If you're using the latest from Git, select:
 * Boards -> Generic nRF51822
 * Soft device -> None
 * Chip -> 32kB RAM, 256 kB flash (xxcc)
 * Programmer -> JLink

The crystal is currently need for the nRF5 branch of the arduino-BLEPeripheral library. However, I'm planning to add a #define in the variant to specify is an external crystal is present.

rogerclarkmelbourne, 2016-07-04:
OK

I will git it another try, but I thought I'd almost done the same thing by copying the pinmap from the generic nRF51822 into the the nRF52 DK variant

I guess I should post a separate question about whether the S130 soft device works with this repo, as AFIK the S132 is for the nRF52 only ??

OSHChip-Owner, 2016-07-04:
Hi Roger,
The low frequency clock selection is done by your application code, as part of initialization of the soft device, with code that might look like this:
   `SOFTDEVICE_HANDLER_INIT(NRF_CLOCK_LFCLKSRC_XTAL_20_PPM, false);`
(taken from a heart rate monitor example)

S130 is for nRF51 , S132 is for nRF52

In other news, I sent you an email back in January, and again yesterday, via your web site contact page, but did not get a reply. Did you receive these emails?

rogerclarkmelbourne, 2016-07-04:
Hi Philip,

I don't seem to have received the emials via the contact page of my site, I'll double check its working. I know the comments emial notifications work but the contacts page may be emailing it somewhere else :-(

Re: Soft device

Thanks

I presumed the S132 was for NRF52. 

Re: Low refreq init. 
I think the problem is more ingrained in my old RBL based repo, there is  code which waits for the 32khz osc to startup.
I tried unsuccessfully to change it, but I didn't make the change to the soft device init handler.

I should probably have tested the startup code without the soft device, but it now looks like this repo will supersede the RBL one that I was using

I'll try doing as Sandeep suggest, ie use the tools from the JSON package and the core from the github clone and see if I can get the LED to flash on my WaveShare nRF51822 board (which has the 32kHz crystal)

rogerclarkmelbourne, 2016-07-04:
Hi Philip
I just tested my website, and there was a screw up after changing the "theme" in the site, which stopped the contact requests getting though to me. The new Theme has its own setup for this, and doesn't use the WordPress default, but we were not aware of this, otherwise we'd have configured it.

Anyway, its working now

So thanks for letting me know this was not working.

rogerclarkmelbourne, 2016-07-04:
Philip

I got you latest contact email and have tried to reply.

OSHChip-Owner, 2016-07-04:
Thanks, received.

sandeepmistry, 2016-07-13:
> I guess I should post a separate question about whether the S130 soft device works with this repo, as AFIK the S132 is for the nRF52 only ??

@rogerclarkmelbourne yes please, I see that you already have #22 open.

I'm closing this for now.

closed 2016-07-13 by sandeepmistry

nihaopaul, 2016-08-16:
@sandeepmistry the open PR from me on both https://github.com/sandeepmistry/arduino-nRF5 and https://github.com/sandeepmistry/arduino-BLEPeripheral added a menu flag to set the LFCLK :)