sandeepmistry/arduino-nRF5 #20: Linker warnings on all boards except nRF52 DK

rogerclarkmelbourne, 2016-07-05:
ld.exe: warning: changing start of section .heap by 4 bytes
ld.exe: warning: changing start of section .stack_dummy by 4 bytes
ld.exe: warning: changing start of section .heap by 4 bytes
ld.exe: warning: changing start of section .stack_dummy by 4 bytes
ld.exe: warning: changing start of section .heap by 4 bytes
ld.exe: warning: changing start of section .stack_dummy by 4 bytes

I presume this isn't important, never the less, ideally something should be changed in the linker file to fix this.

sandeepmistry, 2016-07-13:
@rogerclarkmelbourne any ideas on what we can change?

Agreed, they seem harmless, I believe I've also seen them on the nRF52 with some sketches.

micooke, 2017-09-12:
I believe we are seeing this because of all the . = ALIGN(4) everywhere, and need to add a . = ALIGN(4) at the start of these sections. I haven't resolved this, sorry.

From what I have been reading, and from the lack of things going randomly wrong I think we can ignore this. If you take out `-Wl,--warn-section-align` from `ldflags` in platform.txt it wont show up.