sandeepmistry/arduino-nRF5 #21: Cant get it to work with an nRF51822 board by Waveshare

rogerclarkmelbourne, 2016-07-05:
Unfortunately I can't get this core to work with my nRF51822 board.  (http://www.waveshare.com/nrf51822-eval-kit.htm) http://www.waveshare.com/w/upload/1/1b/BLE400-Schematic.pdf

The board uses an nRF51822 QFAC
It has LED's on pins 18 to 22

I install the JSON package
Cloned the git repo as per the readme

JLink programming doesnt directly work for me so I've been using the hex file generated by the IDE in the tmp folder, uploading via JFlash or the JLink CLI exe

Sketch should just link P0.20

'''
void setup() {
  // put your setup code here, to run once:
  pinMode(20,OUTPUT);
}

void loop() {
    // put your main code here, to run repeatedly:
  digitalWrite(20,HIGH);
  delay(500);
  digitalWrite(20,LOW);
  delay(500);  
}
'''

I tried with and without the S130 softdevice. Neither worked :-(

So I'm kinda stumped at the moment about what to try next.

nihaopaul, 2016-07-06:
try the following, pins dont necessarily match up
```
void setup() {
  // put your setup code here, to run once:
  for(i=0; i<32; i++) {
    pinMode(i,OUTPUT);
  }
}

void loop() {
    // put your main code here, to run repeatedly:
  for(i=0; i<32; i++) {
    digitalWrite(i,HIGH);
  }
  delay(500);
  for(i=0; i<32; i++) {
    digitalWrite(i,LOW);
  }
  delay(500);  
}```

also with the jlink: connect, SWDIO, SWDCLK, VCC( 3.3v) and GND and try again using the programmer then:
```
JLinkExe -device nRF51 -speed 4000 -if SWD
```
to confirm the connection

rogerclarkmelbourne, 2016-07-06:
@nihaopaul 

Thanks

That worked.

I'll now need to figure out the mapping

rogerclarkmelbourne, 2016-07-06:
Umm

I'm not sure why it wasn't working yesterday

As the mapping looks fine today.

I'll see if it still works after I load the S130 softdevice

rogerclarkmelbourne, 2016-07-06:
Its all working now. Thanks

I've actually forked the repo, and I've now made a new variant for the Waveshare BLE400

Which I'll send to Sandeep as a PR

closed 2016-07-06 by rogerclarkmelbourne