sandeepmistry/arduino-nRF5 #22: Generic nRF51822 doesnt work with S130 or S110 softdevices

rogerclarkmelbourne, 2016-07-06:
I'm using the http://www.waveshare.com/product/modules/wireless/bluetooth/nrf51822-eval-kit.htm and have made a new variant, but this problem applies equally to the Generic nRF51822 variant

If I compile the blink all sketch, to flash all pins, and compile without a soft device, it works fine.

However, if I install the S130 softdevice s130_nrf51_2.0.1_softdevice.hex  and select S130 from the menu, the sketch fails to toggle any pins :-(

I've tried an old version of the S130 soft device  s130_softdevice_v9.0.hex (https://developer.nordicsemi.com/nRF5_SDK/nRF51_SDK_v9.x.x/doc/9.0.0/)  but this doesnt work either.

I also tried s110_nrf51_8.0.0_softdevice.hex  but that didn't work

Note. The Jlink interface from the IDE isnt working, so I retrieved the files downloaded by the "nRF5 flash softdevice" command and flashed manually using jFlash. But the files are being flashed to the correct location i.e 0x0000 and seem to be the correct size.

sandeepmistry, 2016-07-13:
Hi @rogerclarkmelbourne,

This might be related to the application address the soft device expects. Or the way your are flashing it.

The linker script is assigned here: https://github.com/sandeepmistry/arduino-nRF5/blob/master/boards.txt#L258

Linker files can be found here: https://github.com/sandeepmistry/arduino-nRF5/tree/master/cores/nRF5/SDK/components/softdevice/s130/toolchain/armgcc - I believe they have the correct addresses.

Tested SoftDevice URL's can be found here: https://github.com/sandeepmistry/arduino-nRF5/blob/master/extras/ide-tools/nRF5FlashSoftDevice.java#L40-L41

* Are you selecting the correct chip variant for your board? (xxaa vs xxac)
* Does your JLink programmer erase all the flash when programming? If so, you should try to merge the IDE built hex with the soft device
* You can try to set a pin in main.cpp prior to init.

rogerclarkmelbourne, 2016-07-13:
Sandeep

I've tried loads of things.

The linker script changes the linked base address of the binary so that it is at 0x1b000 instead of 0x00000 which looks correct for the latest version of the S130 SD as its smaller than the old version (which took just under 0x1c000)

As a test, I tried changing the RBL arduino repo to link to 0x0000 instead of 0x1C000 but their repo doesn't work either if you move the code add from its normal location

I've spoken to Philip (OSHChip) as well about this, and as far as we know the SD should just redirect all interrupts that its not using to the correct location in the sketch code as long as its in the correct location, i.e the same code is supposed to run at 0x00000 and 0x1B000, but it doesn't appear to.

Unfortunately, at the moment I don't have time to investigate this any further.

sandeepmistry, 2016-07-13:
Ok, since it's working for me at this time, and you can't investigate any further, I will close this.

However, feel free to reopen it it if interests you again in the future.

closed 2016-07-13 by sandeepmistry

rogerclarkmelbourne, 2016-07-14:
Hi Sandeep

I think I know whats wrong, though I don't have time to verify it

Your origin appears to be wrong for all your SD linker files

It looks like a typo, as this

```
  RAM (rwx) :  ORIGIN = 0x20002080, LENGTH = 0xdf80
```

Should be
```
  RAM (rwx) :  ORIGIN = 0x20002800, LENGTH = 0xd800
```

https://devzone.nordicsemi.com/question/72218/s130-200-rom-and-ram-settings/

sandeepmistry, 2016-08-24:
@sabas1080 could you please try out @rogerclarkmelbourne's suggest above? (https://github.com/sandeepmistry/arduino-nRF5/issues/22#issuecomment-232532459)

If it solves your S130 soft devices issue from https://github.com/sandeepmistry/arduino-nRF5/issues/30#issuecomment-240499497 please submit a pull request.

sabas1080, 2016-08-25:
Hi @rogerclarkmelbourne @sandeepmistry 

It does not work for me, I try with [this post](https://devzone.nordicsemi.com/blogs/723/rom-and-ram-management/)

keep trying later

rogerclarkmelbourne, 2016-08-26:
I'm pretty sure the Origin is a typo, but there may be other reasons why S130 doesnt work.
If I get chance I'll take a look to see if it works for me

sabas1080, 2016-08-26:
There is an error Typo in

https://github.com/sandeepmistry/arduino-nRF5/blob/master/cores/nRF5/SDK/components/softdevice/s130/toolchain/armgcc/armgcc_s130_nrf51822_xxac.ld#L9

```
RAM (rwx) :  ORIGIN = 0x20002080, LENGTH = 0x5F80
```

for

 ```
RAM (rwx) :  ORIGIN = 0x20002800, LENGTH = 0x5F80
```

but still it does not work for me

rogerclarkmelbourne, 2016-09-02:
I have similar issues with the Nordic SDK11 examples

This could be an issue with the RAM size on your device.

Do you have 16k (QFAA) or 32k (QFAC) nRF51822 devices

I thought my WaveShare board had a 32k device in it, ( http://www.waveshare.com/product/modules/wireless/bluetooth/core51822.htm )  but when I looked at the MCU part number, I found that the ones I originally ordered were 32K but the two additional modules I ordered were only 16k 

I think I may have bought the 16k modules from AliExpress or eBay,  e.g.
http://www.ebay.com.au/itm/NRF51822-2-4G-Wireless-Module-Wireless-BLE4-0-Bluetooth-Communication-Module-/251914914321

and although they look identical to the 32k WaveShare boards, they are only 16k (which is actually shown in the listing)

I presumed WaveShare produced all of these and they only make 32k versions, but perhaps WaveShare really just make the "Evaluation board" to match those modules, (who are made by another company)

BTW. 
I still have issues with SDK11 examples for RHS etc, as they will not work if I compile and try to run on 16k hardware, even if I change the RAM size.

I have a open question with Nordic about this on their forum https://devzone.nordicsemi.com/question/93224/which-sdk11-gcc-examples-work-with-nrf51822/?comment=93336#comment-93336

So perhaps this will shed some light on the problem

sabas1080, 2016-09-03:
I have WaveShare board 32k (QFAC)

I'll see forum, and I'll try

rogerclarkmelbourne, 2016-09-03:
OK
QFAC should be OK, only QFAA has a problem because of low RAM

You could try 

  FLASH (rx) : ORIGIN = 0x1b000, LENGTH = 0x25000
  RAM (rwx) :  ORIGIN = 0x20001870, LENGTH = 0x2790

As these are the settings for SDK11 ble_beacon which works on both QFAA and QFAC

But I don't understand why ORIGIN of RAM is so low

sabas1080, 2016-09-04:
Testing

```
FLASH (rx) : ORIGIN = 0x1b000, LENGTH = 0x25000
  RAM (rwx) :  ORIGIN = 0x20001870, LENGTH = 0x2790
```
Not working for me, test with example "test" of Library BLEPeripheral

rogerclarkmelbourne, 2016-09-04:
OK
I think we'll need to wait for my question on the Nordic forum to be answered as there seem to be some issues with SDK 11 and the nRF51822 (16k) .

rogerclarkmelbourne, 2016-09-06:
I'm not sure if this is applicable, but I got a response from the Nordic forum 
https://devzone.nordicsemi.com/question/93224/which-sdk11-gcc-examples-work-with-nrf51822/?comment=93674#comment-93674

And it looks like the V2 soft devices don't work with all IC versions

> What is the rest of the marking on the 16 kB RAM chips you have? There are 3 generations of QFAA devices, and the latest SDK and SoftDevice are only compatible with the 3rd generation, marked "QFAAH.."

sandeepmistry, 2016-09-07:
So I just flashed my BluzDK board with the Blink example sketch + no soft device, S110, and S130 => all blink successfully. This matches my earlier test results from a few months ago (no changes to the linker files in the master branch).

`nRF51822-QFAC(build code: A1) 256kB Flash` is what OpenOCD reports.

I'll try the RBL BLE Nano v1.0 next, it only has 16k of RAM.

> But I don't understand why ORIGIN of RAM is so low

I think it really depends on how many attributes you use with the soft device. Less services + characteristics + descriptors will need less RAM from my understanding.

rogerclarkmelbourne, 2016-09-07:
Sandeep

The response I had to my question on the Nordic forum, was that the RAM usage does depend on which services are configured.

I'm not sure when those SD options are configured, if its at runtime, then its hard to set that RAM value in a single linker file, unless the value is the worst case for all services enabled.

sandeepmistry, 2016-09-07:
@rogerclarkmelbourne sorry, it was hard for me to keep up with you're activity on the Nordic forum :)

There's definitely some issues with Blink and the RBL BLE Nano v1.0 (`nRF51822-QFAA(build code: G2) 256kB Flash`).

Both S110 and S130 do not work. S110 blinks rapidly, which might indicate a crash. S130 just has solid LED.

> I'm not sure when those SD options are configured, if its at runtime, then its hard to set that RAM value in a single linker file, unless the value is the worst case for all services enabled.

Do you mean something like this? https://github.com/sandeepmistry/arduino-BLEPeripheral/blob/master/nRF51822.cpp#L122-L134

rogerclarkmelbourne, 2016-09-07:
Hi Sandeep

No worries.

I have been posting to the Nordic forum because their examples won't run on QFAA hardware
Nordic claim this is the RAM size, but I'm not sure, as I used the RAM settings from the ble beacon in the hrs but it won't run.

Until the latest posting / reply from Nordic, I wasn't aware of the issue with the V2.0 SD on older nRF51 devices.
I just checked and I also have some QFAAG3 devices - though the ones that still won't run the HRS example are QFAAH

So I'm still unsure about what the issue is with Nordic's examples and armgcc
(Their own hex files, eg. for the HRS,  run fine on the QFAA but the code when compiled with gcc does not run on QFAA

Anyway getting back to the V2.0 SD and devices before QFAAH... 

Personally I don't think I will migrate my repo to SDK 11 or 12, as I'm sure there will be a lot of people who may still have old devices (I bought mine less than a year ago)

But with your repo. I don't know if you can support nRF51 if in reality the V2.0 SD does not support some devices that are over a year old

sandeepmistry, 2016-09-12:
@rogerclarkmelbourne I'm leaning towards only supporting newer QFAA and QFAC for now. If there's more demand for older QFAA h/w we can figure it out later.

sandeepmistry, 2016-09-12:
@sabas1080 have you been able to get blink running with S110 and S130?

What build code does openocd show for your QFAC? 

Note: I tested my BBC micro:bit (QFAA), only S110 worked with blink, S130 did not. I guess it's h/w compatibility issue like @rogerclarkmelbourne mentioned above.

sandeepmistry, 2016-10-02:
@rogerclarkmelbourne  @sabas1080 I think I've figured out the problem S110 and S130 look good on both my RBL Nano v1 and BBC micro:bit.

The soft device flasher was not erasing all for the flash before, so something must have been left in a bad state. I've pushed a change to master to erase all of the flash before flashing a soft device (https://github.com/sandeepmistry/arduino-nRF5/commit/043822aec3723124aa2be8f78f2bd0b6cf8f05f5).

rogerclarkmelbourne, 2016-10-02:
I will re-test, but I was not using your soft device flasher, so I don't know is that's the only issue.

But as I've not tested this for a while, it could well work now.

BTW. In case you are interested, I have the old RBL Arduino repo working with a $20 smartwatch
https://www.youtube.com/watch?v=SBHyQ1OSM-4

But in the longer term I want to get it working with your repo, as the RBL stuff is a dogs dinner, with its layers of mbed and Arduino API etc

sandeepmistry, 2016-10-03:
Sounds good, just make sure you erase all the flash before programming the soft device. I didn't think this was necessary until yesterday.

rogerclarkmelbourne, 2016-10-04:
Hi Sandeep

I've retested and its working fine now. Well the blink works with and without the S130 

I have not however tested any BLE stuff yet.

siddhesh23sawant, 2018-10-15:
hi ! i am trying to program my BLE nRF52832 in arduino software and it is working fine if i didnt select the softdevice (S132) & as i want to load S132 in my BLE, The arduino software lags to upload the code !
please help me out to solve this issue @sandeepmistry