sandeepmistry/arduino-nRF5 #24: Unable to get code on the board

mitko9000, 2016-07-16:
Hi guys I have nrf52 DK board and I would like to use Arduino ide to program it.

But when I try to flash new bootloader or adding softdevice or adding bootloader is leading alywas to errors.

My board have PCA10040 any and its date is prior to 2016.9 

Can someone give me more detailed information how to get started from the scratch with that board and the code available at this repo ?

nihaopaul, 2016-07-17:
What errors are you getting?

mitko9000, 2016-07-17:
Info : only one transport option; autoselect 'swd'
adapter speed: 10000 kHz
cortex_m reset_config sysresetreq
Error: unable to find CMSIS-DAP device
Error: No Valid JTAG Interface Configured.
Error while flashing SoftDevice.

nihaopaul, 2016-07-17:
Can you select under tools > port the NRf52 DK? 

if not, see this revision to see if it applies to you: https://github.com/sandeepmistry/arduino-nRF5/commit/4cc0744b0f0720c64e13ac02a811857adb829d8f

mitko9000, 2016-07-17:
Yes I can select the device com port

nihaopaul, 2016-07-17:
and when you select the NRF52-DK from the port list you can then program it? (not with upload with programmer).

did you try this:

* For boards prior to ```2016.9``` (see sticker), the lastest JLink bootloader is required to upload sketches. To upgrade, press the boot/reset button while powering on the board and copy over the latest [bootloader](https://www.nordicsemi.com/eng/nordic/Products/nRF52-DK/nRF5x-OB-JLink-IF/52275).

to copy it over you just drag and drop the file onto the directory that mounts in your system.

nihaopaul, 2016-07-20:
@mitko9000 is this issue resolved for you now?

mitko9000, 2016-07-21:
No I always get the error above and I am not able neither upload to board nor upload bootloader or softdevice.

When I drag and drop the file inside the system a LED is starting to blink on board and that's going on forever

sandeepmistry, 2016-07-24:
@nihaopaul thanks for helping out!

@mitko9000 please confirm that you have updated the bootloader on the JLink Segger. This must be done outside the Arduino IDE as mentioned by @nihaopaul  in https://github.com/sandeepmistry/arduino-nRF5/issues/24#issuecomment-233184940.

mitko9000, 2016-07-27:
@sandeepmistry My board is PCA10040 / v 0.9.0 / 2015.41

I entered into bootloader mode using the reset button and powering on the board it popped a new  drive named bootloader I drag/drop the .bin file but I get LD5 blinking green forever.

How can I update it otherwise ?

OSHChip-Owner, 2016-07-27:
Hi Miko9000,
There seem to be 3 versions of the firmware released this year:
1)  At Nordic's main web site (J-Link OB-SAM3U128-V2-NordicSemi 160212.bin):
     https://www.nordicsemi.com/eng/nordic/Products/nRF52-DK/nRF5x-OB-JLink-IF/52275

2) Hidden in a Nordic Devzone article (J-Link OB-SAM3U128-V2-NordicSemi 160315.bin):
   https://devzone.nordicsemi.com/question/79002/nrf52-preview-devkit-unable-to-j-link-nrfgo-auto-firmware-update/

3) The latest from Segger , built into their J-Link program (160705)
    Download Seggers V6 software from here:
     https://www.segger.com/downloads/jlink

For the first 2, on your nRF52-DK, slide the power switch to off, hold down the reset button
and while keeping the button down, turn the power back on. This puts it into bootloader mode.
Drag and drop the .BIN file to the drive, and after the transfer is complete, turn the power off and back on again. The drive name should change to JLINK.

For 3 (the most recent version), Install the J-Link software, and start the J-Link program. It will detect
that you do not have the most recent firmware, and offer to update it. Note that for this, you do not have to cycle the power.

While running the J-Link program, you can chec the version of the firmware in your nRF52-DK with the "f" command.  
Here are the results for all 3 versions:

Firmware: J-Link OB-SAM3U128-V2-NordicSemi compiled Feb 12 2016 12:15:53
Firmware: J-Link OB-SAM3U128-V2-NordicSemi compiled Mar 15 2016 18:03:17
Firmware: J-Link OB-SAM3U128-V2-NordicSemi compiled Jul  5 2016 08:42:09

mitko9000, 2016-07-28:
@OSHChip-Owner  Hi I got the file from the seconds link inside the forum and now I see the following device : JLink CDC UART Port.

I can select it inside the Arduino IDE and I click flashSoftDevice but instead going into the jar folder it tries to access -> AppData\Roaming\Arduino15\packages\sandeepmistry\hardware\nRF5\0.1.0/cores/nRF5/SDK/components/softdevice/s132/hex/s132_nrf52_2.0.0_softdevice.hex but inside that folder I have file AppData\Roaming\Arduino15\packages\sandeepmistry\hardware\nRF5\0.1.0/cores/nRF5/SDK/components/softdevice/s132/hex/s132_nrf52_2.0.1_softdevice.hex
So I renamed the file from 2.0.0 to 2.0.1 and it said flashing softdevice done.

After that I tried to upload the empty Arduino sketch on the board and it said -> AppData\Local\Temp\build2212095705496029354.tmp/core.a: No such file or directory
Error compiling.

Any help with that I am suing Arduino 1.6.5

mitko9000, 2016-07-28:
Hi All again,

I downloaded Arduino ide 1.6.9 and I was able to upload code so everything is working.

Can you please tell me is it possible at this stage to do ble things throughout Arduino ?

nihaopaul, 2016-07-29:
great to hear, totally can do: https://github.com/sandeepmistry/arduino-BLEPeripheral 
i use the nrf5 branch

mitko9000, 2016-07-30:
@nihaopaul I found the same repo downloaded the files but I get that the nrf52 DK is not supported if I add only define parameter will that work as expected on the nrf52 ?
Or I have to make changes it did not see can you assist ?

mitko9000, 2016-07-30:
@nihaopaul  Managed to compile the library.
I get this warning 
 warning: changing start of section .heap by 4 bytes but I think its ok as long its only warning

sandeepmistry, 2016-08-14:
@nihaopaul thanks for helping out!

closed 2016-08-14 by sandeepmistry

ahester57, 2017-11-13:
Hello. I know this is old, but on Linux the solution is to add yourself to the 'dialout' group.
Type:  $ sudo usermod -a -G dialout <username>,
where username is, well, your username.
This fixed it for me on Linux.