sandeepmistry/arduino-nRF5 #27: Current repo when installed per instructions does not install all variants

OSHChip-Owner, 2016-07-28:
The current README.md lists 4 supported boards.
The current boards.txt contains entries for 6 boards:
. nRF52DK.name=nRF52 DK
. BluzDK.name=Bluz DK
. BLENano.name=RedBearLab BLE Nano
. RedBearLab_nRF51822.name=RedBearLab nRF51822
. Generic_nRF51822.name=Generic nRF51822
. Waveshare_BLE400.name=Waveshare BLE400

The current package_nRF5_boards_index.json only lists the first 2 boards
The current arduino-nRF5-0.1.0.tar.gz only includes variants files for the first 2 boards

When the instructions in the README.md are followed, only nRF52 DK and Bluz DK are shown in the boards manager.

dlabun, 2016-07-31:
If you download the package from Github as a ZIP it includes all of the boards listed.

OSHChip-Owner, 2016-07-31:
The point is that the README.md, package_nRF5_boards_index.json, and arduino-nRF5-0.1.0.tar.gz all need to be updated to be consistent.

sandeepmistry, 2016-09-12:
I've just released v0.2.0, everything is in sync now.

closed 2016-09-12 by sandeepmistry