sandeepmistry/arduino-nRF5 #29: Unable to flash softdevice (Incorrect file version)

dlabun, 2016-07-29:
When trying to flash the S132 soft device I am getting an error saying it cannot open the HEX file. From looking at the error log OpenOCD is looking for 2.0.0 of the soft device when 2.0.1 is actually in the soft device folder. I was able to rename the 2.0.1 HEX over to 2.0.0 and it flashed successfully.

mitko9000, 2016-07-30:
I had the same problem just renamed the file to the latest version I got from Nordic website

sandeepmistry, 2016-08-22:
@dlabun @mitko9000 do either of you have time to submit a pull request for this change?

Changes need to be made in `boards.txt`. Maybe we can also have a common variable or something to reduce the number of changes needed for future version changes?

dlabun, 2016-08-22:
I can certainly try to submit a pull request, never done one before.

dlabun, 2016-08-22:
Nordic now has v3.0.0 of the S132 soft device available and it's tagged as the "production" version. Should we be targeting that version now?

sandeepmistry, 2016-08-22:
@dlabun sure, we'll have have to update the IDE plugin too: https://github.com/sandeepmistry/arduino-nRF5/blob/master/extras/ide-tools/nRF5FlashSoftDevice.java#L42

Please try v3.0.0 and let us know what the results are. As well as submitting a PR to migrate to it if things look good.

dlabun, 2016-08-22:
Ok, I will update everything to reflect v3.0.0 and report back once I have it working.

mitko9000, 2016-08-22:
@sandeepmistry  Is there anything still that needs to be done ?
 I was away from keyboard for several weeks and I am seeing this now.

dlabun, 2016-08-22:
@sandeepmistry I've just realized two important things... First, this is really a duplicate of #13 and has already been corrected. I didn't see the fix as it hasn't been released yet.

Secondly, moving to v3 of the S132 SD means updating all of the APIs as well. Since this issue isn't related to updating to v3 I'm thinking this issue should be closed and I'll do a PR once I get v3 working.

sandeepmistry, 2016-08-24:
@dlabun good points, I agree. Excited for your SD v3 pull request :)

closed 2016-08-24 by sandeepmistry