sandeepmistry/arduino-nRF5 #30: nRF5FlashSoftDevice support BusPirate?

sabas1080, 2016-08-01:
Hi

I have a Bus Pirate, could you use it to flash?

Thanks

sabas1080, 2016-08-03:
I have compile a version of OpenOCD for Linux 64 with support for Buspirate but i have the next error

`Error: Can't change session's transport after the initial selection was made
Error while flashing SoftDevice.`

My cfg for Buspirate is

`interface buspirate

#source [find interface/buspirate.cfg]

# you need to specify port on which BP lives
buspirate_port /dev/ttyUSB0

# increase working area to 16KB
set WORKAREASIZE 0x4000

reset_config none separate

transport select swd`

sandeepmistry, 2016-08-14:
@sabas1080 what if you supply the configuration to openocd on the command line?

```
openocd -f interface/buspirate.cfg -c "buspirate_port /dev/ttyUSB0; set WORKAREASIZE 0x4000; reset_config none separate; transport select swd"
```

Once you get this working, we can try to incorporate into `programmers.txt` and `platform.txt`

sabas1080, 2016-08-14:
@sandeepmistry from terminal:
[image]

sabas1080, 2016-08-15:
Here program nRF51 with Ruby and Bus Pirate 

http://floe.butterbrot.org/matrix/hacking/nrf/

sabas1080, 2016-08-17:
I 'm uploading firmware compiled in Arduino IDE and Bus Pirate with [programmer](https://github.com/floe/programmer) and softdevice130

`ruby -I. flash.rb --adapter=buspirate,dev=/dev/ttyUSB0 s130_nrf51_2.0.1.bin 0x00000000
`

I 'm trying to blink

`ruby -I. flash.rb --adapter=buspirate,dev=/dev/ttyUSB0 Blink.ino.hex 0x0001C000
`

but I can not get the blink

sandeepmistry, 2016-08-21:
@sabas1080 cool!

Since the Bus Pirate will require Ruby, not sure if this repo will include support for it out of the box. If you're looking for an inexpensive programmer ST-Link clones work well (https://www.adafruit.com/products/2548) also available on eBay, etc.

 Can you try with the S110 soft device? I think your problem might be related to #22. Could you please try adjust the RAM config and continue the discussion there?

sabas1080, 2016-08-22:
Thanks @sandeepmistry 

Working test of the Library [nRFPeripehtal](https://github.com/sandeepmistry/arduino-BLEPeripheral) with S110 Soft

The Instructions are as follows:

Convert .hex to .bin
`arm-none-eabi-objcopy -I ihex -O binary s110_nrf51_8.0.0_softdevice.hex s110_nrf51_8.0.bin`

flash with next command

`ruby -I. flash.rb --adapter=buspirate,dev=/dev/cu.usbserial-xxxxxxxx s110_nrf51_8.0.bin 0x00000000`

Next convert scketch of .hex to .bin
`arm-none-eabi-objcopy -I ihex -O binary test.ino.Waveshare_BLE400.hex test.bin
`
and flash with next command
`
ruby -I. flash.rb --adapter=buspirate,dev=/dev/cu.usbserial-xxxxxxxx Blink.bin 0x00018000`

I am going to buy an STM32 board for try with Black Magic Probe if repo will not support for it out the box

closed 2016-08-22 by sabas1080