sandeepmistry/arduino-nRF5 #31: External libraries do not compile

dlabun, 2016-08-07:
Not sure what's going on with this one, but select external libraries will not compile.... For example if you try to use the Adafruit GFX library it will come back with a "No such file or directory" error. I know this library works perfectly fine if I select any other Arduino board. Same issue occurs with libraries I wrote personally.

sandeepmistry, 2016-08-14:
@dlabun please share the full verbose output from the compile. Also, which IDE version are you using?

dlabun, 2016-08-14:
Hi Sandeep,

I've attached the verbose from the same sketch compiled against your nRF BSP and the stock Arduino Zero BSP. I have tried this with both 1.6.9 and .10 with the same results.

Thanks,
Doug

[nrf52.txt](https://github.com/sandeepmistry/arduino-nRF5/files/417299/nrf52.txt)
[zero.txt](https://github.com/sandeepmistry/arduino-nRF5/files/417300/zero.txt)

sandeepmistry, 2016-08-21:
@dlabun thanks.

Could you please also most the sketch?

dlabun, 2016-08-22:
Something must have changed in 1.6.11 because it's compiling correctly today... Could like have related to [Arduino #5199](https://github.com/arduino/Arduino/pull/5199)?

sandeepmistry, 2016-08-22:
Good question. Not a high priority for me to look into at this point, but let me know if you figure out what the change was.

closed 2016-08-22 by sandeepmistry