sandeepmistry/arduino-nRF5 #32: Question: What is the impact supporting bootloader with DFU?

bojh, 2016-08-09:
At the moment it is possible to select a softdevice and flash it to the BLE enabled device. Now it would be nice to support DFU-Service also - but what is the price? On Nordic [side](http://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk5.v11.0.0%2Fexamples_ble_dfu.html&cp=6_0_0_4_3_1) we can read that a BLE application must be extended for supporting. But in my inial assesment we are not yet prepared on BLE side (BLEPeripheral library). The following changes are required to add BLE DFU Service support to an application:

> The application must use the Device Manager. If the application does not use the Device Manager yet, you must update the application. See BLE Device Manager for more information.
> You must add DFU related files to the BLE application project. See Including DFU files in application.
> You must implement a function that executes application-specific operations to ensure a graceful shutdown of the application. See Implementing graceful shutdown.
> The application must correctly set up the BLE services. See Setting up the BLE services.
> The application must propagate SoftDevice BLE events to the BLE DFU Service. See Propagating BLE stack events.

So in my rough idea:
1. We need the dfu enabled bootloader. May be flushed via Arduino menu like the softdevice way. In the SDK there are prepared bootloader hex files with dfu - like the softdevice hex files in `nRF5_SDK_11.0.0/examples/dfu/bootloader`
2. We have to set and implement a new BLEPeripheral callback for the gracefull shutdown.
3. We have to propagte the BLE events to the deployed bootloader build-in DFU service
and I do not know at the moment what's all done in the nordic Device Manger at all.

In a first (more or less stupid ;-) shot I tried to use mergehex.exe to combine SD+application with that bootloader hexfile - may be to get a hint. For merging only `SD & App.` this is valid, but mergehex does say: `ERROR: The hex files cannot be merged since there are conflicts.`  ... not detailed but seems explicable, because upper preconditions are not given with my standard BLEPeripheral services exteneded application - and I do not have checked the valid addressing ranges.

Does anybody have experience or a deeper knowledge on this area, how to start progress in this direction?

dlabun, 2016-08-09:
You may want to standby until v12 of Nordic SDK is released, a Nordic employee said on their forum that official Arduino support was coming in v12... It's entirely possible DFU support is coming in the new release.

bojh, 2016-08-10:
OK, this would also be a good option.  But we have to wait therefore. There are known any roadmap dtates? Yes, and I saw, that Arduino want come with an officall nRF52832 powered board: Arduino Primo, in cooperation with [Nordic](https://www.nordicsemi.com/eng/News/News-releases/Product-Related-News/Arduino-Primo-base-board-features-native-Bluetooth-low-energy-wireless-connectivity-and-NFC-Touch-to-Pair-using-Nordic-Semiconductor-nRF52832-SoCs). With that, the (Io-)Things makes sense.

dlabun, 2016-08-10:
Nordic did not release any specific dates, all he said was "soon". If I see any sort of dates I'll post them here.

 I saw the Arduino,org Primo as well, but I am suspicious of it. Their website is bragging how it's now available for purchase yet if you ask about it on their forum it's "still under development" and not available for purchase.

sandeepmistry, 2016-08-21:
No plans to support DFU in this project.

I would suggest asking RedBear to open source their bootloader from https://github.com/RedBearLab/nRF51822-Arduino

closed 2016-08-21 by sandeepmistry