sandeepmistry/arduino-nRF5 #34: Servo Library Doesn't Work?

alexwchen, 2016-08-13:
Hi Everyone, 

Before I start, I just want to thank Sandeep and everyone who contribute to this library, it is incredible work :) 

I am trying to get servo to work on the nrf52 board. I notice on the bottom of this github page it says we are using the core based on the Arduino SAMD Core. However, when I run the Servo Sweep Example. It Give me the following message. 

--------------------
WARNING: library Servo claims to run on [avr architecture(s) and may be incompatible with your current board which runs on sam architecture(s).
In file included from /var/folders/13/7vdzws8x02d0r7rmwljm676c0000gn/T/arduino_modified_sketch_940886/Sweep.ino:10:0:
/Applications/Arduino.app/Contents/Java/libraries/Servo/src/Servo.h:69:2: error: #error "This library only supports boards with an AVR, SAM or SAMD processor."
 #error "This library only supports boards with an AVR, SAM or SAMD processor."
  ^
exit status 1
Error compiling for board nRF52 DK.
--------------------

I dig further into the Servo.h and find out there is this condition in the code
---------------
// Architecture specific include
#if defined(ARDUINO_ARCH_AVR)
#include "avr/ServoTimers.h"
#elif defined(ARDUINO_ARCH_SAM)
#include "sam/ServoTimers.h"
#elif defined(ARDUINO_ARCH_SAMD)
#include "samd/ServoTimers.h"
#else
#error "This library only supports boards with an AVR, SAM or SAMD processor."
#endif
---------------

I am wondering since we are SAMD, maybe it should pass the conditioning and select SAMD? Or the SAMD we are using here is different than what they are looking for in the code. 

Thank you so much, please let me know if there is anything unclear about my question :) Will be happy to clearify. 

Thanks again, 

Alex

sandeepmistry, 2016-08-14:
Hi @alexwchen,

The servo library will have to be updated with a nRF51/nRF52 timer implementation. Even though both the SAMD and nRF51/52 have M0 ARM cores, they have totally different ways to enable timers.

I'm going to close this because there's nothing that can be done in this core, the Servo library needs changes in the end.

closed 2016-08-14 by sandeepmistry

carlosperate, 2017-03-09:
Hi @sandeepmistry,

Could the RedBear Servo library (https://github.com/RedBearLab/nRF51822-Arduino/tree/S130/arduino-1.6.x/hardware/RBL/RBL_nRF51822/libraries/Servo) be easily ported and added to the libraries directory?
I only needed for the nRF51, but ideally it'd be great to submit a PR here so that it's available for all the cores/variants. Are there significant differences in the timers between the two cores? There are differences for the wiring_analog, so just wondering how much effort it might take.

dlabun, 2017-03-09:
There's was a discussion in another issue thread and we came to the conclusion that this core is really intended to get the basic nRF hardware features working. Something like the Servo library or Bluetooth would remain a standalone library... Of course that's always subject to change.

sandeepmistry, 2017-03-13:
@carlosperate it looks feasible. I think the timer registers are the same.

@dlabun the other option is to submit a PR to the official Arduino Servo lib. What do you think?

carlosperate, 2017-03-14:
Thanks for the clarification @dlabun and @sandeepmistry.

Maybe you have already discussed this, but most of the other board packages I've seen usually include the servo library as well, as it does depend on platform specific peripherals. If a PR was to be submitted to the official Arduino Servo library, the nrf5 platform would have to be included alongside the avr, sam, and samd; which seems quite out of place, no?
I guess it could be it's own independent library, but if it needs this specific core to work (as other nrf packages might clash with their timer usege), then why separate them?

sandeepmistry, 2017-03-14:
@carlosperate good point!

The official Servo lib has a fixed set of architectures in the library.properties (avr,sam,samd). So, like you said it makes sense to bundle an nrf5 specific on with this core.

carlosperate, 2017-03-18:
I've got something working on an nrf51822, I can probably submit a PR next week after I find some time to clean it up and do a few more tests, but don't really have any nRF52 dev boards around to test it.

sandeepmistry, 2017-03-19:
@carlosperate awesome!

I'm sure we can get some nRF52 boards tested, I'll have to check if I have a servo at home to test with.

kartik9k, 2018-08-03:
Did this PR ever get merged? Just wanted to tinker with Servo on nrf52.

Great work regardless! :D

anant3110, 2018-08-23:
The Servo.h has been updated since the issue was raised and the updated conditions are :
```
#if defined(ARDUINO_ARCH_AVR)
#include "avr/ServoTimers.h"
#elif defined(ARDUINO_ARCH_SAM)
#include "sam/ServoTimers.h"
#elif defined(ARDUINO_ARCH_SAMD)
#include "samd/ServoTimers.h"
#elif defined(ARDUINO_ARCH_STM32F4)
#include "stm32f4/ServoTimers.h"
#elif defined(ARDUINO_ARCH_NRF52)
#include "nrf52/ServoTimers.h"
#else
#error "This library only supports boards with an AVR, SAM, SAMD, NRF52 or STM32F4 processor."
#endif
```
However, the code still does not compile correctly and raises the "This library only supports boards with an AVR, SAM, SAMD, NRF52 or STM32F4 processor" error message. Is the NRF52 device used here different from what @sandeepmistry has written this library for or is ARDUINO_ARCH_NRF52 not defined? 

Additionally, is there any other Servo library which I can work with on my NRF52 device (Redbear BLE Nano 2) ?

shwetankshrey, 2018-08-23:
The Servo.h has been updated since the issue was raised and the updated conditions are :
```
#if defined(ARDUINO_ARCH_AVR)
#include "avr/ServoTimers.h"
#elif defined(ARDUINO_ARCH_SAM)
#include "sam/ServoTimers.h"
#elif defined(ARDUINO_ARCH_SAMD)
#include "samd/ServoTimers.h"
#elif defined(ARDUINO_ARCH_STM32F4)
#include "stm32f4/ServoTimers.h"
#elif defined(ARDUINO_ARCH_NRF52)
#include "nrf52/ServoTimers.h"
#else
#error "This library only supports boards with an AVR, SAM, SAMD, NRF52 or STM32F4 processor."
#endif
```
However, the code still does not compile correctly and raises the "This library only supports boards with an AVR, SAM, SAMD, NRF52 or STM32F4 processor" error message. Is the NRF52 device used here different from what @sandeepmistry has written this library for or is ARDUINO_ARCH_NRF52 not defined? 

Additionally, is there any other Servo library which I can work with on my NRF52 device (Redbear BLE Nano 2) ?

kiter4ev, 2018-09-22:
@shwetankshrey try Arduino primo , it has working nrf52 servo library