sandeepmistry/arduino-nRF5 #35: Binary Integers not recongized

dlabun, 2016-08-22:
If you try to declare an integer in the binary format it will fail to compile... Verbose is attached, cheesy example sketch below.

`int Test = B001;

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}`
[zero.txt](https://github.com/sandeepmistry/arduino-nRF5/files/429214/zero.txt)
[nrf.txt](https://github.com/sandeepmistry/arduino-nRF5/files/429219/nrf.txt)

sandeepmistry, 2016-08-22:
`cores/arduino/binary.h` does not match the SAMD core.

https://github.com/arduino/ArduinoCore-samd/blob/master/cores/arduino/binary.h

Would you mind trying to submit a pull request to make if the same as the SAMD core?

dlabun, 2016-08-22:
That's a nifty little file... I'll take care of the pull request.

sandeepmistry, 2016-09-11:
Fixed via #42.

closed 2016-09-11 by sandeepmistry