sandeepmistry/arduino-nRF5 #38: Error while flashing SoftDevice.

AmittSharma, 2016-09-01:
Hi,
I am trying to flash my Bluzdk by following the steps given by you. But as I proceed to the last step i.e. 

Select Tools -> nRF5 Flash SoftDevice
Read license agreement
Click "Accept" to accept license and continue, or "Decline" to decline and abort
If accepted, SoftDevice binary will be flashed to the board

I get the following error:

java.io.IOException: Cannot run program "/home/amitt/.arduino15/packages/sandeepmistry/tools/openocd/0.10.0-dev.nrf5/bin/openocd": error=2, No such file or directory
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at java.lang.Runtime.exec(Runtime.java:620)
	at java.lang.Runtime.exec(Runtime.java:485)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:11)
	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:129)
	at cc.arduino.packages.uploaders.SerialUploader.burnBootloader(SerialUploader.java:410)
	at com.github.sandeepmistry.arduino.nRF5.nRF5FlashSoftDevice.lambda$run$0(nRF5FlashSoftDevice.java:143)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.io.IOException: error=2, No such file or directory
	at java.lang.UNIXProcess.forkAndExec(Native Method)
	at java.lang.UNIXProcess.<init>(UNIXProcess.java:248)
	at java.lang.ProcessImpl.start(ProcessImpl.java:134)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 7 more
Error while flashing SoftDevice.

Please help, how do I proceed now ?

dlabun, 2016-09-01:
Two questions: Did you install the BSP per the instructions and what version of the soft device are you trying to flash? There is a known bug where 2.0.1 is downloaded but the BSP is hardcoded to look for 2.0.0

nihaopaul, 2016-09-02:
also install via the board manager and not the master branch for an easier life!

dlabun, 2016-09-02:
You need to follow the directions in the Readme as it contains steps for setting up to flash soft devices.

sandeepmistry, 2016-09-05:
@nihaopaul @dlabun thanks for helping out!

@AmittSharma please see the suggestions above, and re-open the issue if they don't work for you.

closed 2016-09-05 by sandeepmistry

tlonch, 2017-02-28:
Hi,
I have the same error than @AmittSharma when trying to flash softdevice, even after applied the suggestions above. I work on Linux with the nRF52DK and use s132 v.2.0.1.

Could you help me please ?

sandeepmistry, 2017-03-02:
@tlonch please try https://github.com/sandeepmistry/arduino-nRF5/issues/101#issuecomment-282485964.

tlonch, 2017-03-02:
Thank for your reply. So I installed `libc6:i386` and `libstdc++6:i386`. I no longer have last error but get this one : 
`/home/tim/.arduino15/packages/sandeepmistry/tools/openocd/0.10.0-dev.nrf5/bin/openocd: error while loading shared libraries: libudev.so.1: cannot open shared object file: No such file or directory`

So i installed `libudev1:i386` and `libudev0:i386`.

After flashing softdevice I obtain : 
`Open On-Chip Debugger 0.10.0-dev-00254-g696fc0a (2016-04-09-12:07)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 0
0
adapter speed: 10000 kHz
cortex_m reset_config sysresetreq
nrf52.cpu: target state: halted
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x000008e4 msp: 0x20000400
** Programming Started **
auto erase enabled`

Does this seem good to you ? Sorry if I am asking stupid questions, I am no developer and Linux is new for me

sandeepmistry, 2017-03-03:
@tlonch output looks cutoff but the portion you shared looks good. Does the IDE report success?

Could you please submit a PR to let other linux users know what extra packages they need to install? Thanks.

tlonch, 2017-03-04:
You 're right  i forgot a portion. Full report : 
>Open On-Chip Debugger 0.10.0-dev-00254-g696fc0a (2016-04-09-12:07)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 0
0
adapter speed: 10000 kHz
cortex_m reset_config sysresetreq
nrf52.cpu: target state: halted
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x000008e4 msp: 0x20000400
** Programming Started **
auto erase enabled
wrote 114688 bytes from file /home/tim/.arduino15/packages/sandeepmistry/hardware/nRF5/0.2.0/cores/nRF5/SDK/components/softdevice/s132/hex/s132_nrf52_2.0.1_softdevice.hex in 29.118017s (3.846 KiB/s)
** Programming Finished **
** Verify Started **
verified 110636 bytes in 1.763829s (61.255 KiB/s)
** Verified OK **
** Resetting Target **
shutdown command invoked

I made the PR, hope it's ok. Thanks for the help @sandeepmistry  !

sandeepmistry, 2017-03-13:
Just to follow up @tlonch opened PR #113.

Carbon18, 2020-07-30:
> Two questions: Did you install the BSP per the instructions and what version of the soft device are you trying to flash? There is a known bug where 2.0.1 is downloaded but the BSP is hardcoded to look for 2.0.0

Hi @dlabun - This is my exact issue at this point. I finally have the SoftDevice S130 changed to 2.0.1 in the softdevices.txt, but I don't see a README file. Any help getting to the next steps would be appreciated.

Erickrk, 2020-08-25:
Thanks a lot for the help, @Carbon18! I was having this issue on Linux so I changed the file accordingly to what you said, it indeed had a link that differs from the filename right under it.  
Even with this, the problem kept happening,  So I decided to move back to Windows where I had a very similar issue, solved it by copying the .hex file of the nrf52_2.0.1_softdevice to the hex folder in AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.6.0\cores\nRF5\SDK\components\softdevice\s132.
Still don't know what happened in Linux though.

LakshayNegi108, 2023-03-09:
> Thanks a lot for the help, @Carbon18! I was having this issue on Linux so I changed the file accordingly to what you said, it indeed had a link that differs from the filename right under it. Even [...]

what's the next step after copying the hex file in s132? Arduino flash soft Device still gives the same error.