sandeepmistry/arduino-nRF5 #39: Will this become the official core for the Arduino Primo ?

rogerclarkmelbourne, 2016-09-04:
I've just realised that the Arduino Primo will use the nRF52 as the main processor, and as Sandeep works for Arduino, I wondered if this repo will be the Core that the Primo uses, or whether Nordic are developing a separate core for the Repo.
In which case it seems pointless continuing with this one, as an official core written by Nordic developers would probably be better ??

sabas1080, 2016-09-05:
Arduino Primo is of Arduino.org or not?

Sandeep works for Arduino.cc

rogerclarkmelbourne, 2016-09-05:
Sorry. My mistake.

I will contact Arduino.org about this

closed 2016-09-05 by rogerclarkmelbourne

sandeepmistry, 2016-09-07:
This repo is maintained independently of either Arduino in my spare time. 

> In which case it seems pointless continuing with this one, as an official core written by Nordic developers would probably be better ??

I'm not sure what you mean't by "better" here. However, it's impossible to compare the quality of a project that is in the public domain to something that is not released.

I'm going to continue maintaining this version for my own use in either case.

rogerclarkmelbourne, 2016-09-07:
Sorry

I was under the initial impression that Nordic were writing a core for Arduino.org

I have since contacted Arduino.org and have been informed that they are writing the core themselves with support from Nordic.
i.e its not Nordic who are writing the core.

It will be interesting to what Arduino.cc publish, as my understanding the nRF52 core will be released in a few weeks (1 or 2 weeks)