sandeepmistry/arduino-nRF5 #40: NRF_RTCx

nihaopaul, 2016-09-06:
Has anyone been able to use the RTC, i noticed the delay.c using it but havnt been able to use it.

`
static unsigned irq_counter = 0;
boolean LD = false;

void RTC0_IRQHandler(void) {
  irq_counter++;
  if (irq_counter % 8 == 0) {
    LD = !LD;
    digitalWrite(9, false);
  }
  NRF_RTC0->EVENTS_TICK = 0UL;
}

void stopRTC() {
  NRF_RTC0->TASKS_STOP = 0UL;
}
void setRTC() {
  //stop to set options
  stopRTC();
  
  NVIC_SetPriority(RTC0_IRQn, 15);
  NVIC_ClearPendingIRQ(RTC0_IRQn);
  NVIC_EnableIRQ(RTC0_IRQn);
  
  NRF_CLOCK->LFCLKSRC = (uint32_t)((CLOCK_LFCLKSRC_SRC_RC << CLOCK_LFCLKSRC_SRC_Pos) & CLOCK_LFCLKSRC_SRC_Msk);
  NRF_CLOCK->TASKS_LFCLKSTART = 1UL;
  
  NRF_RTC0->PRESCALER = (1<<12) -1;
  NRF_RTC0->EVTENSET = offsetof(NRF_RTC_Type, EVENTS_TICK);
  NRF_RTC0->INTENSET = offsetof(NRF_RTC_Type, EVENTS_TICK);
  NRF_RTC0->TASKS_START = 1UL;
}

void setup() {
  pinMode(9,OUTPUT);
  digitalWrite(9, LD);
  setRTC();
}

void loop() {
  __WFI();
}
`

sandeepmistry, 2016-09-06:
@nihaopaul maybe you need to do this, because the .ino gets translated to a .cpp file:

```c++
extern "C"
{
  void RTC0_IRQHandler(void) {
    // ...
  }
}
```

OSHChip-Owner, 2016-09-06:
If you are doing this with the Soft Device included (S130 or S132), RTC0 is not available. See the list of blocked and restricted peripherals in the appropriate SD specification.

dlabun, 2016-09-07:
Just a heads up, there is a documented issue with the nRF52 RTC that requires a workaround to get it to work correctly... There's more info on the Nordic website [http://infocenter.nordicsemi.com/topic/com.nordic.infocenter.nrf52832.Rev1.errata/err_832_new.html](url)

rogerclarkmelbourne, 2016-09-07:
Yikes

I was just about to buy some nRF52 modules, but I suspect they will have devices in them which are 3 to 6 months old (possibly more)

I guess I should hold off buying a nRF52 device until they fix all the issues shown on that page

dlabun, 2016-09-07:
It's not all that bad, most of the issues have software work arounds. Personally I'm waiting for the nRF52 modules made by Dynastream since they "own" ANT and are deeply involved with Nordic's development of the nRF series.

rogerclarkmelbourne, 2016-09-07:
OK.

Perhaps I should just buy 1 and then buy another one in the future when these issues are resolved correctly.

sandeepmistry, 2016-09-26:
Closing this for now, I think https://github.com/sandeepmistry/arduino-nRF5/issues/40#issuecomment-244910249 is the answer.

closed 2016-09-26 by sandeepmistry