sandeepmistry/arduino-nRF5 #41: BLE example

nestorsilk, 2016-09-11:
Arduino-nRF5 work very well for me. Using it. with a nrf52 development kit. succeeded also in changing the path structure within the platform.txt to map the sdk structure as a valid Arduino path structure. But when trying to start BLE examples I ran into a lot of troubles as it is difficult to find out which SDK 11 files are really necessary to run within the Arduino environment.

Therefore I wanted to ask whether someone already succeeded to implement a BLE example with Arduino-nRF5 for the nRF52832 MCU?

Tom

sabas1080, 2016-09-11:
Hi

I am using library BLE Peripheral branch nRF5
https://github.com/sandeepmistry/arduino-BLEPeripheral/tree/nRF5

for development in nRF51 and examples work

nestorsilk, 2016-09-11:
Thank you very much. Was not aware of the existence of this library. Works like a charm. Excellent!

closed 2016-09-12 by nestorsilk

sandeepmistry, 2016-09-12:
@sabas1080 thanks for helping out!

I've added a section for BLE in the read me (https://github.com/sandeepmistry/arduino-nRF5#ble).

Also, the `BLEPeripheral` nRF5 branch has been merged to master, and a new v0.3.0 version should be available in the Arduino IDE library manager in the next hour or so.