sandeepmistry/arduino-nRF5 #43: Rationale behind this core?

joseangeljimenez, 2016-09-12:
Hello,

This is a question about the main differences between this core (sandeepmistry) and the one by RedBearLab. Setting apart some obvious differences: this core supports also the nRF52 chipset, may I ask what are the main philosophical reasons (rationale) behing this core?

I kindly suggest you may also add this rationale to the the main README.md of the repository. It will be of great help for newcomers trying to understand the different approaches to developing BLE chipsets.

I am a seasoned C++ developer with some experience in the low-level programming of the nrF51 series. It is great to see the effort you are putting together in this core. Thank you!

This core:
https://github.com/sandeepmistry/arduino-nRF5

RedBearLab core:
https://github.com/RedBearLab/nRF51822-Arduino

sandeepmistry, 2016-09-17:
Hi @joseangeljimenez,

> Setting apart some obvious differences: this core supports also the nRF52 chipset, may I ask what are the main philosophical reasons (rationale) behing this core?

 * main reason is nRF52 support
 * there is also no custom boot loader needed
 * the RedBearLab core is also based on mbed, which at the time I received my board did not support nRF52. This adds some overhead/complexity and issues like https://github.com/RedBearLab/nRF51822-Arduino/issues/46.

> I kindly suggest you may also add this rationale to the the main README.md of the repository. It will be of great help for newcomers trying to understand the different approaches to developing BLE chipsets

Do you mean something like: "Program you nRF51 or nRF52 board using the Arduino IDE"?

Let me know.

Thanks for the feedback!

joseangeljimenez, 2016-09-17:
Thanks a lot for the explanation. I had missed the bootloader differences and also the fact that RBL is reusing/mixing classes from both frameworks (mbed and Arduino).

"Program you nRF51 or nRF52 board using the Arduino IDE" is a nice motto, but I find much more informative your small summary in the three bullet points above. 👍 

Kudos!

sandeepmistry, 2016-09-26:
I've updated the read me in https://github.com/sandeepmistry/arduino-nRF5/commit/a60bc7becf438ffacb554bbbe9e92d35f74802c7.

I'll hold of mentioning my bullet points in the read me for now, I'm not too keen on comparing differences with the RBL core.

closed 2016-09-26 by sandeepmistry