sandeepmistry/arduino-nRF5 #45: Why does this core need GCC 5.2 ?

rogerclarkmelbourne, 2016-10-04:
Is there any specific reason why gcc 5.2 has to be installed ?

From what I've read on the Nordic blog they don't recommend using gcc thats this new

I think the last thing I read, as that gcc 4.9.3 was the recommended version 
See https://devzone.nordicsemi.com/tutorials/7/development-with-gcc-and-eclipse/

`**Note - 12-3-2016 **

Was made aware of a potential bug that may have been introduced in the latest 5.2 release by @Lucas in the comments. Problem is described in this thread. Will look into this, but for now you may want to use the 4.9.3 release to be safe. `

Also see

https://devzone.nordicsemi.com/question/70735/nrf52-pca10040-debugging-fails-eclipse/

which recommends 4.9

rogerclarkmelbourne, 2016-10-04:
Actually, I think I'll close this as perhaps the very latest Gcc releases fixes the issue

closed 2016-10-04 by rogerclarkmelbourne

sandeepmistry, 2016-10-04:
> I think I may have asked this question, and I think the answer is utoa is needed from 5.2, but just to note, that

It was just the latest at the time, debugger support if not a big deal for me at the moment.

> Actually, I think I'll close this as perhaps the very latest Gcc releases fixes the issue

Are you thinking the GCC should be upgraded to 5.4? https://launchpad.net/gcc-arm-embedded/+download

rogerclarkmelbourne, 2016-10-04:
I tried 5.4 2016 q2 and it seemed to work, though I'm only testing by blinking some LEDs

The only problem I had was that I installed it in the arduino tools folder under arm-none-eabi-gcc and strangely the Arduino IDE seems to use whatever the latest version of GCC is inside that folder, i.e if you have multiple verisons of gcc inside arm-none-eabi-gcc  it ignore the older ones and just uses the folder with the highest version number.

I think if you are using a more recent version than 4.9.3, then you may as well go for the last stable release i.e 5.4 2016q2 (or perhaps q3 when its released)

sandeepmistry, 2016-10-05:
Ok, I'm not in a rush to update to the latest GCC unless there's a specific reason to do so.

>  Arduino IDE seems to use whatever the latest version of GCC is inside that folder, i.e if you have multiple verisons of gcc inside arm-none-eabi-gcc

If you have something like this in all the core's, it should select the specific version: https://github.com/sandeepmistry/arduino-nRF5/blob/master/platform.txt#L30

rogerclarkmelbourne, 2016-10-06:
OK.
Thanks