sandeepmistry/arduino-nRF5 #50: EEPROM simulation and Flash access.

d00616, 2016-10-10:
I have implemented accessing internal flash for nrf51 MCU for RIOT-OS. https://github.com/RIOT-OS/RIOT/pull/2239

My idea is to make this available for arduino and set up a EEPROM simulation using last two Flash pages.

sandeepmistry, 2016-10-10:
Please open a pull request when your library is ready.

sandeepmistry, 2016-11-20:
Closing this for now do to lack of activity.

https://github.com/cmaglie/FlashStorage/pull/7 might be of interest.

closed 2016-11-20 by sandeepmistry

d00616, 2016-11-20:
I'm working on porting official EEPROM library with my own emulation using last 8k of FLASH allowing 15,000,000 write operations for an single cell or 15,000 writes for complete emulated 1k EEPROM.

I have seen FlashStorage. I think it's incompatible with 1k/4k page sizes.