sandeepmistry/arduino-nRF5 #51: printf support

d00616, 2016-11-01:
At the moment i play around with printf. This doesn't work until i have modified Print.cpp like https://forum.arduino.cc/index.php?topic=197601.msg1488213#msg1488213

I think this is not a perfect solution because file is ignored. I have no idea about a correct implementation.

diff --git a/cores/nRF5/Print.cpp b/cores/nRF5/Print.cpp
index 05a1f18..71f7551 100644
--- a/cores/nRF5/Print.cpp
+++ b/cores/nRF5/Print.cpp
@@ -37,6 +37,16 @@ size_t Print::write(const uint8_t *buffer, size_t size)
   return n;
 }
 
+// for printf compatibility
+extern "C" {
+__attribute__((weak))
+int _write(int file, char *buffer, int size)
+{
+  Serial.write(buffer, size);
+       return 0;
+}
+}
+
 size_t Print::print(const __FlashStringHelper *ifsh)
 {
   return print(reinterpret_cast<const char *>(ifsh));

sandeepmistry, 2016-11-06:
Hi @d00616,

> This doesn't work until i have modified Print.cpp like

I think this is fine, it's the same behaviour as the official Arduino SAMD core. Also, `printf` is not really "Arduino" style.

> I think this is not a perfect solution because file is ignored.

What do you mean by this?

sandeepmistry, 2016-11-20:
Closing for now due to lack of activity.

closed 2016-11-20 by sandeepmistry

d00616, 2016-11-20:
Is a pull request with this solution welcome?

sandeepmistry, 2016-11-22:
Not at this time, the current goal is to track the official SAMD core. It requires the same workaround to make `printf` functional.