sandeepmistry/arduino-nRF5 #52: Crash on ISR

d00616, 2016-11-05:
This code is hanging after first timer/rtc event on NRF51 and NRF52. I think this code is correct.

`//#define TIMER
#define RTC

#include "nrf.h"

uint16_t counter = 0;

void setup() {
  Serial.begin(38400);

  NRF_CLOCK->EVENTS_HFCLKSTARTED = 0;
  NRF_CLOCK->TASKS_HFCLKSTART = 1;

  while (NRF_CLOCK->EVENTS_HFCLKSTARTED == 0) {}

  NRF_TIMER2->MODE = TIMER_MODE_MODE_Timer;
  NRF_TIMER2->PRESCALER = 2;

  #ifdef TIMER
    NVIC_SetPriority(TIMER2_IRQn, 15);
    NVIC_ClearPendingIRQ(TIMER2_IRQn);
    NVIC_EnableIRQ(TIMER2_IRQn);
  #endif
  NRF_TIMER2->BITMODE = TIMER_BITMODE_BITMODE_16Bit;
  NRF_TIMER2->CC[0] = 4000;
  NRF_TIMER2->INTENSET = TIMER_INTENSET_COMPARE0_Enabled << TIMER_INTENSET_COMPARE0_Pos;
  NRF_TIMER2->TASKS_START = 1;

  #ifdef RTC
    NVIC_SetPriority(RTC0_IRQn, 15);
    NVIC_ClearPendingIRQ(RTC0_IRQn);
    NVIC_EnableIRQ(RTC0_IRQn);
  #endif
  NRF_RTC0->PRESCALER = 32;
  NRF_RTC0->EVTENSET = RTC_EVTENSET_COMPARE0_Msk;
  NRF_RTC0->INTENSET = RTC_INTENSET_COMPARE0_Msk;
  NRF_RTC0->CC[0] = 2000;
  NRF_RTC0->TASKS_START = 1;

}

void loop() {
  Serial.print("Counter: ");
  Serial.println(counter);
  delay(300);
}

void TIMER_2_ISR(void)
{
  counter++;
  NRF_TIMER2->EVENTS_COMPARE[0] = 0;
#if __CORTEX_M == 0x04
    volatile uint32_t dummy = NRF_TIMER2->EVENTS_COMPARE[0];
    (void)dummy;
#endif
}

void RTC0_IRQHandler(void)
{
  counter++;
  NRF_RTC0->EVENTS_COMPARE[0] = 0;

#if __CORTEX_M == 0x04
    volatile uint32_t dummy = NRF_RTC0->EVENTS_COMPARE[0];
    (void)dummy;
#endif
}`

sandeepmistry, 2016-11-06:
@d00616 

Are you sure it's crashing?

I think you are missing the `extern "C"` clauses around the ISR's:

```
extern "C" {

void TIMER_2_ISR(void)
{
  counter++;
  NRF_TIMER2->EVENTS_COMPARE[0] = 0;
#if __CORTEX_M == 0x04
    volatile uint32_t dummy = NRF_TIMER2->EVENTS_COMPARE[0];
    (void)dummy;
#endif
}

void RTC0_IRQHandler(void)
{
  counter++;
  NRF_RTC0->EVENTS_COMPARE[0] = 0;

#if __CORTEX_M == 0x04
    volatile uint32_t dummy = NRF_RTC0->EVENTS_COMPARE[0];
    (void)dummy;
#endif
}

}
```

d00616, 2016-11-07:
Hello @sandeepmistry,

thank you for the hint. At the moment there is a bug whith 'extern "C"' -> https://github.com/arduino/arduino-builder/pull/128

'Extern "C"' is only working with all code in one line.

Thank you.

closed 2016-11-07 by d00616

cheeta1, 2017-05-10:
hi @d00616  have you been able to run the above code for timer 2 interrupt on nrf51822? if yes then can you share the code?

d00616, 2017-05-10:
@cheeta1 You have to put all interrupt code into one line:

```
#include <nrf.h>

#define RTC NRF_RTC0

int interrupt = 0;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println("Start");

  RTC->TASKS_STOP = 1;
  RTC->PRESCALER = 31; //1024Hz frequency
  RTC->CC[0] = RTC->COUNTER + (3 * 1024);
  RTC->EVTENSET = RTC_EVTENSET_COMPARE0_Msk;
  RTC->INTENSET = RTC_INTENSET_COMPARE0_Msk;
  RTC->TASKS_START = 1;
  RTC->EVENTS_COMPARE[0] = 0;
}

void loop() {
  // put your main code here, to run repeatedly:
  Serial.print(millis());
  Serial.print(" ");
  Serial.print(RTC->COUNTER);
  Serial.print(" ");
  Serial.println(RTC->EVENTS_COMPARE[0]);
  Serial.print(" ");
  Serial.println(interrupt);
  delay(1000);
}

extern "C" { void RTC0_IRQHandler(void) { interrupt++; }}

```

cheeta1, 2017-05-11:
[image]

cheeta1, 2017-05-11:
hi @d00616  i got the above output at serial which shows that ISR is not called. can you tell mw why the ISR is not being called?

d00616, 2017-05-13:
I have updated the example code. On NRF51 NVIC_EnableIRQ is required. The NRF5_RESET_EVENT helps to handle resetting the event on NRF51 and NRF52.