sandeepmistry/arduino-nRF5 #57: Missing file error after fresh install

Btje1977, 2016-11-13:
I followed the install procedure with a new arduino installal and the additional board package.

Arduino: 1.6.12 (Windows 10), Board:"Generic nRF51822, 16 kB RAM, 256 kB flash (xxaa), None, Crystal Oscillator"

In file included from C:\Users\BramenRianne\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.2.0\cores\nRF5/Arduino.h:5:0,

                 from sketch\sketch_nov13a.ino.cpp:1:

c:\users\bramenrianne\appdata\local\arduino15\packages\sandeepmistry\tools\gcc-arm-none-eabi\5_2-2015q4\lib\gcc\arm-none-eabi\5.2.1\include\stdint.h:9:26: fatal error: stdint.h: No such file or directory

compilation terminated.

exit status 1
Fout bij het compileren van board Generic nRF51822

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Btje1977, 2016-11-13:
When compiling for a standard arduino bord it is ok

Btje1977, 2016-11-13:
The file is there

c:\users\bramenrianne\appdata\local\arduino15\packages\sandeepmistry\tools\gcc-arm-none-eabi\5_2-2015q4\lib\gcc\arm-none-eabi\5.2.1\include\

sandeepmistry, 2016-11-14:
@Btje1977 I'm not a full-time Windows user, but maybe the path is too long? What filesystem type is you C:\ drive?

Could you also enable verbose logging in the IDE and provide the output?

Btje1977, 2016-11-14:
File system is ntfs

Arduino: 1.6.12 (Windows 10), Board:"Generic nRF51822, 16 kB RAM, 256 kB flash (xxaa), None, Crystal Oscillator"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\BramenRianne\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\BramenRianne\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\BramenRianne\Documents\Arduino\libraries -fqbn=sandeepmistry:nRF5:Generic_nRF51822:chip=xxaa,softdevice=none,lfclk=lfxo -ide-version=10612 -build-path C:\Users\BRAMEN~1\AppData\Local\Temp\arduino_build_505334 -warnings=all -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.gcc-arm-none-eabi.path=C:\Users\BramenRianne\AppData\Local\Arduino15\packages\sandeepmistry\tools\gcc-arm-none-eabi\5_2-2015q4 -prefs=runtime.tools.openocd.path=C:\Users\BramenRianne\AppData\Local\Arduino15\packages\sandeepmistry\tools\openocd\0.10.0-dev.nrf5 -verbose C:\Users\BRAMEN~1\AppData\Local\Temp\untitled508793661.tmp\sketch_nov14a\sketch_nov14a.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\BramenRianne\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\BramenRianne\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\BramenRianne\Documents\Arduino\libraries -fqbn=sandeepmistry:nRF5:Generic_nRF51822:chip=xxaa,softdevice=none,lfclk=lfxo -ide-version=10612 -build-path C:\Users\BRAMEN~1\AppData\Local\Temp\arduino_build_505334 -warnings=all -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.gcc-arm-none-eabi.path=C:\Users\BramenRianne\AppData\Local\Arduino15\packages\sandeepmistry\tools\gcc-arm-none-eabi\5_2-2015q4 -prefs=runtime.tools.openocd.path=C:\Users\BramenRianne\AppData\Local\Arduino15\packages\sandeepmistry\tools\openocd\0.10.0-dev.nrf5 -verbose C:\Users\BRAMEN~1\AppData\Local\Temp\untitled508793661.tmp\sketch_nov14a\sketch_nov14a.ino
Using board 'Generic_nRF51822' from platform in folder: C:\Users\BramenRianne\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.2.0
Using core 'nRF5' from platform in folder: C:\Users\BramenRianne\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.2.0
Build-opties gewijzigd, alles wordt opnieuw gebuild
Detecting libraries used...
"C:\Users\BramenRianne\AppData\Local\Arduino15\packages\sandeepmistry\tools\gcc-arm-none-eabi\5_2-2015q4/bin/arm-none-eabi-g++" -mcpu=cortex-m0 -mthumb -c -g -Os -w  -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions  -w -x c++ -E -CC -DF_CPU=16000000 -DARDUINO=10612 -DARDUINO_GENERIC -DARDUINO_ARCH_NRF5  -DNRF51 -DNRF5 "-IC:\Users\BramenRianne\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.2.0\cores\nRF5/SDK/components/toolchain/" "-IC:\Users\BramenRianne\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.2.0\cores\nRF5/SDK/components/toolchain/CMSIS/Include" "-IC:\Users\BramenRianne\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.2.0\cores\nRF5/SDK/components/toolchain/gcc/" "-IC:\Users\BramenRianne\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.2.0\cores\nRF5/SDK/components/device/" "-IC:\Users\BramenRianne\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.2.0\cores\nRF5/SDK/components/drivers_nrf/delay/" "-IC:\Users\BramenRianne\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.2.0\cores\nRF5/SDK/components/softdevice/none/headers/" -DUSE_LFXO "-IC:\Users\BramenRianne\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.2.0\cores\nRF5" "-IC:\Users\BramenRianne\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.2.0\variants\Generic" "C:\Users\BRAMEN~1\AppData\Local\Temp\arduino_build_505334\sketch\sketch_nov14a.ino.cpp" -o "nul"
"C:\Users\BramenRianne\AppData\Local\Arduino15\packages\sandeepmistry\tools\gcc-arm-none-eabi\5_2-2015q4/bin/arm-none-eabi-g++" -mcpu=cortex-m0 -mthumb -c -g -Os -w  -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions  -w -x c++ -E -CC -DF_CPU=16000000 -DARDUINO=10612 -DARDUINO_GENERIC -DARDUINO_ARCH_NRF5  -DNRF51 -DNRF5 "-IC:\Users\BramenRianne\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.2.0\cores\nRF5/SDK/components/toolchain/" "-IC:\Users\BramenRianne\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.2.0\cores\nRF5/SDK/components/toolchain/CMSIS/Include" "-IC:\Users\BramenRianne\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.2.0\cores\nRF5/SDK/components/toolchain/gcc/" "-IC:\Users\BramenRianne\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.2.0\cores\nRF5/SDK/components/device/" "-IC:\Users\BramenRianne\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.2.0\cores\nRF5/SDK/components/drivers_nrf/delay/" "-IC:\Users\BramenRianne\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.2.0\cores\nRF5/SDK/components/softdevice/none/headers/" -DUSE_LFXO "-IC:\Users\BramenRianne\AppData\Local\Arduino1 [... cut ...]
In file included from C:\Users\BramenRianne\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.2.0\cores\nRF5/Arduino.h:5:0,

                 from C:\Users\BRAMEN~1\AppData\Local\Temp\arduino_build_505334\sketch\sketch_nov14a.ino.cpp:1:

c:\users\bramenrianne\appdata\local\arduino15\packages\sandeepmistry\tools\gcc-arm-none-eabi\5_2-2015q4\lib\gcc\arm-none-eabi\5.2.1\include\stdint.h:9:26: fatal error: stdint.h: No such file or directory

compilation terminated.

exit status 1
Fout bij het compileren van board Generic nRF51822

ShawnHymel, 2016-11-14:
I'm running into the same issue, too. I'm using Windows 7 and get the exact same error. The best I can figure is that the file _...\AppData\Local\Arduino15\packages\SparkFun\tools\gcc-arm-none-eabi\5_2-2015q4\lib\gcc\arm-none-eabi\5.2.1\include\stdint.h_ has the following:

`#ifndef _GCC_WRAP_STDINT_H
#if __STDC_HOSTED__
# if defined __cplusplus && __cplusplus >= 201103L
#  undef __STDC_LIMIT_MACROS
#  define __STDC_LIMIT_MACROS
#  undef __STDC_CONSTANT_MACROS
#  define __STDC_CONSTANT_MACROS
# endif
# include_next <stdint.h>
#else
# include "stdint-gcc.h"
#endif
#define _GCC_WRAP_STDINT_H
#endif
`

The line `#include_next <stdint.h>`, from what I understand, is trying to include the header file "stdint.h" from another location (likely an OS-wide installation of the gcc-arm toolchain). On Windows, that simply doesn't exist, unless I've somehow installed the gcc-arm toolchain previously.

sandeepmistry, 2016-11-15:
Based on this: http://forum.osdev.org/viewtopic.php?p=242219&sid=719218c9952a817817ee08e717b465ec#p242219

`-ffreestanding` is missing from the compiler flags. Can you try adding it to the end of the `compiler.c.flags=...` and `compiler.cpp.flags=...` lines in `platform.txt`?

ShawnHymel, 2016-11-15:
Looks like stdint.h was found, but it introduced a whole host of new problems: 
```
In file included from C:\Users\shawn.hymel\Documents\Arduino\libraries\stlport/stl/config/features.h:94:0,

                 from C:\Users\shawn.hymel\Documents\Arduino\libraries\stlport/stl/_cprolog.h:18,

                 from C:\Users\shawn.hymel\Documents\Arduino\libraries\stlport/stdlib.h:24,

                 from C:\Users\shawn.hymel\AppData\Local\Arduino15\packages\SparkFun\hardware\nRF5\0.2.1\cores\nRF5/Arduino.h:6,

                 from C:\Users\SHAWN~1.HYM\AppData\Local\Temp\build5cf65f44b05a03dc83f9d469262c20a1.tmp\sketch\Blink.ino.cpp:1:

C:\Users\shawn.hymel\Documents\Arduino\libraries\stlport/stl/config/_system.h:196:4: error: #error Unknown platform !!

 #  error Unknown platform !!

    ^

In file included from C:\Users\shawn.hymel\Documents\Arduino\libraries\stlport/stl/config/_system.h:202:0,

                 from C:\Users\shawn.hymel\Documents\Arduino\libraries\stlport/stl/config/features.h:94,

                 from C:\Users\shawn.hymel\Documents\Arduino\libraries\stlport/stl/_cprolog.h:18,

                 from C:\Users\shawn.hymel\Documents\Arduino\libraries\stlport/stdlib.h:24,

                 from C:\Users\shawn.hymel\AppData\Local\Arduino15\packages\SparkFun\hardware\nRF5\0.2.1\cores\nRF5/Arduino.h:6,

                 from C:\Users\SHAWN~1.HYM\AppData\Local\Temp\build5cf65f44b05a03dc83f9d469262c20a1.tmp\sketch\Blink.ino.cpp:1:

C:\Users\shawn.hymel\Documents\Arduino\libraries\stlport/stl/config/stl_mycomp.h:33:2: error: #error "Your compiler version is not recognized by STLport. Please edit <stlport/stl/config/stl_mycomp.h>"

 #error "Your compiler version is not recognized by STLport. Please edit <stlport/stl/config/stl_mycomp.h>"

  ^

In file included from C:\Users\shawn.hymel\Documents\Arduino\libraries\stlport/stl/_cprolog.h:18:0,

                 from C:\Users\shawn.hymel\Documents\Arduino\libraries\stlport/stdlib.h:24,

                 from C:\Users\shawn.hymel\AppData\Local\Arduino15\packages\SparkFun\hardware\nRF5\0.2.1\cores\nRF5/Arduino.h:6,

                 from C:\Users\SHAWN~1.HYM\AppData\Local\Temp\build5cf65f44b05a03dc83f9d469262c20a1.tmp\sketch\Blink.ino.cpp:1:

C:\Users\shawn.hymel\Documents\Arduino\libraries\stlport/stl/config/features.h:175:6: error: #error "can't determine endianess"

 #    error "can't determine endianess"

      ^

In file included from C:\Users\shawn.hymel\AppData\Local\Arduino15\packages\SparkFun\hardware\nRF5\0.2.1\cores\nRF5/Arduino.h:6:0,

                 from C:\Users\SHAWN~1.HYM\AppData\Local\Temp\build5cf65f44b05a03dc83f9d469262c20a1.tmp\sketch\Blink.ino.cpp:1:

C:\Users\shawn.hymel\Documents\Arduino\libraries\stlport/stdlib.h:50:45: fatal error: ../include/stdlib.h: No such file or directory

compilation terminated.
```

mirceadaneliuc, 2016-11-15:
[image]
@sandeepmistry as you can see in the attachment, the only platform.txt with nrf51822 is the one in the RBL folder which was created when I've previously tried Roger Clarke's method. 

In which file should I try to add the -ffreestanding? 

Thanks!

Mircea

Btje1977, 2016-11-15:
I read somewhere that you are using a different gcc. Is that right and could this be the cause.

Btje1977, 2016-11-15:
I added the lines but no change

Btje1977, 2016-11-16:
Why are u using a different GCC?

Btje1977, 2016-11-16:
Maybe you can have a look at this https://redbearlab.github.io/arduino/package_redbearlab_index.json implementation and how they did it.

Btje1977, 2016-12-01:
Could we use the default GCC?

sandeepmistry, 2016-12-05:
@Btje1977 please try it out, and report your findings ...

dlabun, 2016-12-18:
What board are you using? I did not see these issues with the Nordic nRF52 Dev Kit.

Btje1977, 2016-12-18:
A nrf51822

dlabun, 2016-12-20:
@ShawnHymel Are you using this core or are you using one from Sparkfun? The error messages you posted are all for a Sparkfun core I cannot locate.

Btje1977, 2016-12-20:
I get the error with just an empty ino.

dlabun, 2016-12-20:
@Btje1977 That's to be expected, I'm trying to weed out the issue right now.

dlabun, 2016-12-20:
@Btje1977 I am not able to replicate the issue on Windows 7 so I need to get a Windows 10 install up and running. In the mean time, did try reinstalling the core?

dlabun, 2016-12-22:
@Btje1977 I still cannot replicate this issue on Windows 10. I would recommend you try upgrading to IDE 1.6.13 and reinstalling this core.

sandeepmistry, 2016-12-22:
@dlabun maybe a long username is the trigger? `BramenRianne` and `shawn.hymel` are what have been reported above, how long was the one you tested with?

dlabun, 2016-12-22:
It's entirely possible, so far I've only tested with a default Win10 install. Normally Win10 truncates the user folder name so you don't run into issues. I still have a few more variables to test out.

sandeepmistry, 2016-12-22:
@dlabun good to know, thanks for trying to narrow this down!

dlabun, 2016-12-23:
So it doesn't appear to be a long name issue... I created a user in Win10 with a 16 character name and everything compiled fine. 

@Btje1977 It would be help to know if you made any modifications to the core, how you installed the core and what other cores you have installed. I would like to replicate your setup as closely as possible.

ShawnHymel, 2016-12-23:
@dlabun it was the SparkFun board, so that's good to know. Thanks!

Btje1977, 2016-12-23:
No other boards installed.

I think it goes wrong at the installation It hangs at step 2/2. If i then kill arduino because it doesn't quit and try to install the baord again it is succesfull. My guess is that it doesn't install everything then.

When installing the esp8266 boards all goes well.

Btje1977, 2016-12-23:
I tried it again but first removed the esp and nrf lib again. Now it installed and the blink example compiled OK. Still wierd. The first time I tried it installed ok and then it didn't work. But ok it installed now so I can start developing. Thanx for the help.

What is a good design example for sending  uart strin from an android phone to the ble400

dlabun, 2016-12-23:
It does take some time to install the compiler tools on Windows... Not sure why but even I see that long delay with a high power workstation.

Btje1977, 2016-12-24:
Thanx for the help.

dlabun, 2016-12-24:
Closing as resolved

closed 2016-12-24 by dlabun

sandeepmistry, 2016-12-24:
> It does take some time to install the compiler tools on Windows... Not sure why but even I see that long delay with a high power workstation.

@dlabun it take the IDE a long time to extract the ~100 MB GCC archive, even on Mac. We could look into removing stuff from the `gcc-arm-none-eabi-5_2-2015q4-20151219` archives to speed installation up.

dlabun, 2016-12-24:
@sandeepmistry I am not familiar with how the IDE unpacks files, it is possible to improve the IDE's speed? I've noticed the SAMD core is a bit slow as well.

Btje1977, 2016-12-24:
The esp8266 package installs faster. Is the different gcc needed?

dlabun, 2016-12-24:
I don't think it's an issue of GCC version, both packages have ~100MB tool chains but ESP8266 just unpacks faster. I am wondering if this more a matter of type of compression used.

dlabun, 2016-12-24:
I've opened a new issue so we can keep things cleaner than discussing two issues in one thread.

sandeepmistry, 2017-01-03:
For others to reference, the new issue is: https://github.com/sandeepmistry/arduino-nRF5/issues/71