sandeepmistry/arduino-nRF5 #58: Power consumption seems high

danyocom, 2016-11-15:
Power consumption of an blank app is 4ma (on BLENano and others)... this seems high. Any tips/tweaks for lower power consumption? For comparison, pre-compiled samples from the NRF SDK  use lest than 1/2 of that even when advertising via ble etc.

rogerclarkmelbourne, 2016-11-15:
Blank is different an app that advertises.

If you look at those, you will see they use the power management to put the processor into one of its low power modes between when it advertises

You could do the same thing with blank by adding the appropriate code

sandeepmistry, 2016-11-20:
Thanks @rogerclarkmelbourne!

>If the UARTS, Timers and SPI etc are powered up by default in the startup code, you'd also need to shut them down to save power. even when using a blank app

UART is only enabled on `Serial.begin(..)`, there's s RTC timer setup for every millisecond to support the `millis()` and `delay(...)` API's.

@danyocom did you get a chance to look at Roger's suggestions?

sandeepmistry, 2016-12-05:
Closing for now, due to lack of activity.

closed 2016-12-05 by sandeepmistry