sandeepmistry/arduino-nRF5 #59: Support for internal hardware via external/internal libraries

d00616, 2016-11-19:
The nRF5 Platform provides a lot of internal Hardware like:
- Watchdog Timer
- PPI
- RADIO
- Temperature Sensor
- NFC
- I2S
- Hardware Random Number Generator
....

I think the best way is to provide bundled libraries supporting the internal hardware instead of forking an external library like https://github.com/sandeepmistry/arduino-nRF5/pull/55#issuecomment-261719099 or https://github.com/sandeepmistry/arduino-nRF5/pull/54#issuecomment-258714787

For other platforms a lot of libraries are included:
- [Arduino for AVR](https://github.com/arduino/Arduino/tree/master/hardware/arduino/avr/libraries) comes with "EEPROM", "SPI", "HID", "SoftwareSerial", "Wire".
- [Arduino for SAMD](https://github.com/arduino/ArduinoCore-samd/tree/master/libraries) is bundled with "HID", "SAMD_AnalogCorrection", "SPI", "USBHost" and "Wire".
- [Arduino for ESP8266](https://github.com/esp8266/Arduino/tree/master/libraries) comes with 24 libraries supporting Hard- and Software features.

I think there is no problem to bundle more libraries than Arduino for AVR. This makes it easier for many people use the internal Hardware of nRF5 MCU.

Example: In the future arduino-nRF5 is included into "Board Manager" per default like a lot of other boards. The installation is easy. Nobody sees what is written in README. The most people have no knowledge about complete functionality provided by nRF5 hardware.

To use an external library you must know that the MCU provides a functionality, you have to know that there are external libraries providing Nordic specific functionality and last but not least you have to choose a correct version. Libraries like [SleepyDog](https://github.com/adafruit/Adafruit_SleepyDog/blob/master/library.properties) are marked as compatible but they aren't.

In my opinion, when hardware functionality is implemented in external libraries this results in projects using more hardware, like a external temerature sensor instead of the internal one, or insecure cryptogaphy by seeding a pseudo random number generator with input of an analog pin instead of using random numbers generated with included hardware RNG...

If you wanna provide the hardware functionality across external libraries, i think the best way is to create a github organization which collects all additional libraries. This helps people with advanced knowledge to find out what is implemented. I think in most cases a fork of libraries like SlepyDog or TrueRandom must be maintained by people working with arduino-nrf5.

clowrey, 2016-11-19:
I'd have to agree with this approach.. having as much nrf5x hardware support bundled as possible makes it easy for the beginner (if that is who you are targeting) to get started and make the most of this hardware.

dlabun, 2016-11-21:
Hate to disagree, but I don't think it's a good idea... The nRF chips are highly customizable and bundling the hardware drivers may drive code size through the roof. The radio has quite a few different operating modes and each needs a lot of Nordic SDK code to operate.

d00616, 2016-11-21:
Hello @dlabun,

> The nRF chips are highly customizable and bundling the hardware drivers may drive code size through the roof.

The code for most customizable periphery is includes via SPI, Wire and UART. The code for components like HWRNG, Temperature Sensor doesn't need customization.

Is it better, when everyone has to read the data sheet and has to write it's own drivers?

> The radio has quite a few different operating modes and each needs a lot of Nordic SDK code to operate.

If you mean with mode using a Software Stack like BLE, ANT or ShockBurst, i agree. These are Software features.
I you mean it isn't possible to to send data without any higer protocol between two nRF5 chips i cannot agree. A working implementation is [available here](https://github.com/RIOT-OS/RIOT/tree/master/tests/driver_nrfmin).

dlabun, 2016-11-21:
@d00616 

I am strictly against bundling everything into a single core... I agree SPI, I2C and UART should be in the core as the most basic features that should be available. These features don't depend on the Nordic soft devices, only the radio features do. Really it should be any radio feature that requires a soft device should have it's own library with the appropriate soft device image ready to go.

sandeepmistry, 2016-11-26:
I agree with @dlabun, for now the goal of this project is to provide a similar set of capabilities as the official Arduino SAMD core of which it is based.

From my perspective the ESP8266 core bundles more libraries than it should.

Also, when if comes to library discoverability, bundling then with the core only solves one problem. It's still key to have proper documentation. For example, for the SPI and Wire libraries included in the core the official Arduino reference can be used: https://www.arduino.cc/en/Reference/Libraries because the API is 100% compatible.

> In the future arduino-nRF5 is included into "Board Manager" per default like a lot of other boards.

I'm skeptical this will happen, only official Arduino products are included in the board manager by default.

> If you wanna provide the hardware functionality across external libraries, i think the best way is to create a github organization which collects all additional libraries. This helps people with [...]

Feel free to start the Github organization. I don't have the bandwidth to take on such efforts at this time.

closed 2016-11-26 by sandeepmistry

OwenBrotherwood, 2017-04-09:
Having tried the solution to aid _porting_ (failed due to libraries) my main thought is that the arduino-nRF5 is a distraction.
It is an interesting thought to _introduce_ arduino users to the nRF5 but it is quickly clear that any thoughts of it making easy for beginners is not the actual aim for the repro: the challenge of allowing the nRF5 to be done in Arduino has been achieved.
The repro is not for making the nRF5 easy for beginners (there are much better solutions out there) but allowing Arduino centric non-beginners to take a look at the nRF5.