sandeepmistry/arduino-nRF5 #61: flashing micro:bit

MartinRobotz, 2016-11-20:
I tried to flash the softdevice onto the micro:bit, but get 128 errors when flashing softdevice or Blink sketch:
Segger firmware is installed on microbit, Zadig replaced the JLink driver for WinUSB, 
COM-Port is greyed out in the arduino IDE; 
How can the microbit be flashed from within the Arduino IDE? Anyone who succeeded?  

error report:
cortex_m reset_config sysresetreq
adapter speed: 1000 kHz
nrf51.cpu: target state: halted
target halted due to debug-request, current mode: Thread 
xPSR: 0xc1000000 pc: 0x000006d0 msp: 0x000007c0
** Programming Started **
auto erase enabled
wrote 93184 bytes from file C:\Users\Martin\AppData\Local\Arduino15\packages\sandeepmistry\hardware\nRF5\0.2.0/cores/nRF5/SDK/components/softdevice/s110/hex/s110_nrf51_8.0.0_softdevice.hex in 27.661337s (3.290 KiB/s)
** Programming Finished **
** Verify Started **
jaylink: SWD I/O operation failed: 06.
Error: jaylink_swd_io() failed: JAYLINK_ERR_DEV.
jaylink: Failed to send data to device: LIBUSB_ERROR_TIMEOUT.
jaylink: Failed to send data to device: LIBUSB_ERROR_TIMEOUT.
jaylink: Sending data to device timed out.
jaylink: transport_write() failed: -4.
Error: jaylink_swd_io() failed: JAYLINK_ERR_TIMEOUT.
Error: checksum mismatch - attempting binary compare
jaylink: Last read operation left 1661 bytes.
jaylink: Failed to send data to device: LIBUSB_ERROR_PIPE.
jaylink: transport_write() failed: -1.
Error: jaylink_swd_io() failed: JAYLINK_ERR.
jaylink: Last read operation left 1730 bytes.
jaylink: Failed to send data to device: LIBUSB_ERROR_IO.
jaylink: transport_write() failed: -1.
Error: jaylink_swd_io() failed: JAYLINK_ERR.
diff 0 address 0x00000000. Was 0x60 instead of 0xc0
diff 1 address 0x00000001. Was 0xb8 instead of 0x07

sandeepmistry, 2016-11-22:
> Zadig replaced the JLink driver for WinUSB,

Did you also do this for the CMSIS-DAP programmer?

Can you try again but select: `Tools -> Programmer -> CMSIS-DAP` before flashing.

Note: I've only tested the process from my Mac, not using Windows.

MartinRobotz, 2016-11-23:
@sandeepmistry 
programmer is not working in IDE, 
solution is to drag&drop the output hex-file to the Microbit-drive. REquires to look after the compiled hex in the AppData-Temp directory
Tested a small Serial.print sketch and Serial MOnitor showed the Message.
Looking forward to use the BlePeripheral library after flashing the SoftDevice

MartinRobotz, 2016-11-23:
@sandeepmistry 
yes, with CMSIS-DAP it works. The microbit is flashed over USB with the normal mbed-USB-DAP-driver.
The SoftDevice also is flashed, without any verification dialogue "decline or accept".
Very nice project, looking forward to use the bleperipheral library. Thank you!

sandeepmistry, 2016-11-26:
Ok, great - since it's working I'm going to close this for now.

Maybe you can submit a pull request for a read me change to mention CMSIS DAP should be selected for the micro:bit. Thanks.

closed 2016-11-26 by sandeepmistry