sandeepmistry/arduino-nRF5 #62: Bluetooth Mesh

danlsk, 2016-11-23:
Will this core development roadmap include the Bluetooth Mesh at some point? Currently, there is Nordic Open Mesh for N51/N52. Will this be more immediate one?

sandeepmistry, 2016-11-26:
No plans to include this. My preference would be to have this as an external Arduino Library that is installable via the Arduino IDE Library manager. This would make it possible to also use with other nRF51/52 Arduino cores.

closed 2016-11-26 by sandeepmistry

probonopd, 2018-02-11:
Seems to be implemented in S140,S132, S130 and S110 SoftDevices

Source:
https://www.nordicsemi.com/eng/Products/Bluetooth-low-energy/nRF5-SDK-for-Mesh

probonopd, 2018-05-20:
https://github.com/NordicPlayground/nRF51-ble-bcast-mesh is talking about Mesh and Arduino. Usable for our purpose?