sandeepmistry/arduino-nRF5 #64: BMD-300/BMD-301 Support

Spegs21, 2016-12-01:
I don't have my eval board yet but I've been working on adding support for this module.

I believe the NRF52DK board will probably work for BMD-300 Eval board but I'd also like to support the Rigado bootloader, RigDFU. It has two update options, serial and BLE (Mac/Linux Only). So essentially there are three ways to update this module.

Initially I wanted to include an upload menu under a BMD-300 board where you could select openocd, RigDFU Serial, or RigDFU BLE but I couldn't figure out how to get that to play nice with the platform.txt. So for now I've just been trying to get the serial working.

So far in boards.txt I have:

```
BMD-300/301.name=BMD-300/301 RigDFU

BMD-300/301.upload.tool=sandeepmistry:rigdfu
BMD-300/301.upload.protocol=
BMD-300/301.upload.maximum_size=524288

BMD-300/301.build.mcu=cortex-m4
BMD-300/301.build.f_cpu=16000000
BMD-300/301.build.board=BMD-300
BMD-300/301.build.core=nRF5
BMD-300/301.build.variant=BMD-300_RigDFU
BMD-300/301.build.variant_system_lib=
BMD-300/301.build.extra_flags=-DNRF52
BMD-300/301.build.float_flags=-mfloat-abi=hard -mfpu=fpv4-sp-d16
BMD-300/301.build.ldscript=nrf52_xxaa.ld

BMD-300/301.menu.softdevice.none=None
BMD-300/301.menu.softdevice.none.softdevice=none

BMD-300/301.menu.softdevice.s132=S132
BMD-300/301.menu.softdevice.s132.softdevice=s132
BMD-300/301.menu.softdevice.s132.softdeviceversion=2.0.1
BMD-300/301.menu.softdevice.s132.upload.maximum_size=409600
BMD-300/301.menu.softdevice.s132.build.extra_flags=-DNRF52 -DS132 -DNRF51_S132
BMD-300/301.menu.softdevice.s132.build.ldscript=armgcc_s132_nrf52832_xxaa.ld

BMD-300/301.menu.lfclk.lfxo=Crystal Oscillator
BMD-300/301.menu.lfclk.lfxo.build.lfclk_flags=-DUSE_LFXO
BMD-300/301.menu.lfclk.lfrc=RC Oscillator
BMD-300/301.menu.lfclk.lfrc.build.lfclk_flags=-DUSE_LFRC
BMD-300/301.menu.lfclk.lfsynt=Synthesized
BMD-300/301.menu.lfclk.lfsynt.build.lfclk_flags=-DUSE_LFSYNT
```

And in platform.txt I added:

```
tools.rigdfu.path={runtime.platform.path}/tools/rigdfu
tools.rigdfu.cmd=py
tools.rigdfu.cmd.windows=py.exe

tools.rigdfu.genimage.pattern="{cmd}" -2 {path}/image-tools/genimage/genimage.py {build.path}/{build.project_name}.hex -o {build.path}/{build.project_name}.RigDFU.bin

tools.rigdfu.upload.params.verbose=-d2
tools.rigdfu.upload.params.quiet=-d0
tools.rigdfu.upload.pattern="{cmd}" -3 {path}/update-tools/serial/dfu.py {build.path}/{build.project_name}.RigDFU.bin -s {serial.port}
```

I also added a variant folder. The problem with this is we need to run the .hex through their genimage.py script first before uploading. I couldn't figure out how to get that to work. Maybe add it into the compiler patterns?

I got the scripts from their github which you need to request access to. You need python 2 and 3 so I used the Python Launcher. BLE.py operates in a similar fashion for BLE updates. Encryption options are also available through the DFU.py and BLE.py which would be nice to incorporate.

sandeepmistry, 2016-12-05:
> Maybe add it into the compiler patterns?

I suggest you read: https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification

Closing this for now, this is really board specific and not sure how it really fits in with the goal of this project.

closed 2016-12-05 by sandeepmistry

Spegs21, 2016-12-05:
Already read it. Not dead set on the RigDFU stuff but what is the harm in adding support for the BMD-300 if anything additional may be needed outside of the RigDFU options? It's a varient like any other board.

sandeepmistry, 2016-12-05:
@Spegs21 please submit a pull request once you have the board and have tested with it.