sandeepmistry/arduino-nRF5 #65: Running on Raspberry Pi (OpenOCD issue)

carlosperate, 2016-12-02:
Hi,

I'm trying to run this on a raspberry pi, so I had to manually install the arm gcc (even though we are already on ARM I read that it might still be required for correct compilation for cortext M) and OpenOCD.

Now for OpenOCD I had to download the source code (https://github.com/sandeepmistry/openocd-code-nrf5/) and install from master branch:

```
./bootstrap
./configure --disable-werror
make
sudo make install
```

(a combination from following other instructions and https://github.com/sandeepmistry/openocd-code-nrf5/blob/nrf5/.travis.yml)

So I am currently getting the following error when trying to upload to a micro:bit:

```
Arduino: 1.6.13 (Linux), Board: "BBC micro:bit, None"

Sketch uses 2,988 bytes (1%) of program storage space. Maximum is 262,144 bytes.
Open On-Chip Debugger 0.10.0-dev-00255-g57a19d4 (2016-12-02-00:58)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 0
Error: The specified debug interface was not found (cmsis-dap)
The following debug interfaces are available:
1: ftdi
2: usb_blaster
3: jlink
4: vsllink
the selected serial port 3: jlink
 does not exist or your board is not connected
5: ulink
6: hla
7: osbdm
8: opendous
9: sysfsgpio
10: aice

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

Do I need to install anything else? Any help would be greatly appreciated.

carlosperate, 2016-12-02:
If I try to configure with --enable-cmsis-dap I get the following error:

```
checking for LIBUSB0... no
checking for HIDAPI... no
checking for HIDAPI... no
checking for HIDAPI... no
configure: error: hidapi is required for the CMSIS-DAP Compliant Debugger
```

So I guess I need to try to get hidapi on the Pi.
Is there anything I might need to chase down?

sandeepmistry, 2016-12-02:
Hey @carlosperate, I suggest you take a look at these scripts: https://github.com/arduino/OpenOCD-build-script - in particular [arch.arm.build.bash](https://github.com/arduino/OpenOCD-build-script/blob/master/arch.arm.build.bash).

I think your missing the libusb0, libusb and hid api dev headers. The scripts linked above build those from scratch ....

carlosperate, 2016-12-02:
That sounds like a good step forward, thanks @sandeepmistry!
I'll try again this evening. If I get this compiled in the raspberry pi, are there any steps to package it the way the arduino board manager expects to install it? (is that what the travis script from your fork is doing?) Would be great to be able to add ARM as another supported platform.

carlosperate, 2016-12-02:
So I've run the arch.arm.build.bash script with your fork as ath OpenOCD folder and seems to include CMSIS-PAP in the build:

```
OpenOCD configuration summary
--------------------------------------------------
MPSSE mode of FTDI based devices        yes (auto)
Segger J-Link JTAG Programmer           yes (auto)
ST-Link JTAG Programmer                 yes (auto)
TI ICDI JTAG Programmer                 yes (auto)
Keil ULINK JTAG Programmer              yes (auto)
Altera USB-Blaster II Compatible        yes (auto)
Versaloon-Link JTAG Programmer          yes (auto)
OSBDM (JTAG only) Programmer            yes (auto)
eStick/opendous JTAG Programmer         yes (auto)
Andes JTAG Programmer                   yes (auto)
USBProg JTAG Programmer                 no
Raisonance RLink JTAG Programmer        no
Olimex ARM-JTAG-EW Programmer           no
CMSIS-DAP Compliant Debugger            yes (auto)
```
So it compiles fine, no errors, so I've changed the platform.txt file to the new "object directory":
```
tools.openocd.path=/home/pi/microbit/build-openOCD/objdir/
tools.openocd.cmd=bin/openocd
```
But when trying to use it with the Arduino IDE I get:

```
Arduino: 1.6.13 (Linux), Board: "BBC micro:bit, None"

Sketch uses 2,988 bytes (1%) of program storage space. Maximum is 262,144 bytes.
Open On-Chip Debugger 0.10.0-dev-00255-g57a19d4 (2016-12-02-19:48)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 0
cortex_m reset_config sysresetreq
adapter speed: 1000 kHz
Error: unable to find CMSIS-DAP device
Error: No Valid JTAG Interface Configured.
Error: No Valid JTAG Interface Configured.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
````

sandeepmistry, 2016-12-05:
@carlosperate do you have another non-Raspbery PI Linux PC to test the micro:bit with?

The openocd seemed fine with an x86 bit based Linux PC.

sandeepmistry, 2016-12-18:
Closing this for now due to lack of activity.

closed 2016-12-18 by sandeepmistry

carlosperate, 2016-12-21:
Sorry, I haven't had a chance to test this again.

>@carlosperate do you have another non-Raspbery PI Linux PC to test the micro:bit with?
>
>The openocd seemed fine with an x86 bit based Linux PC.

I'm sure it will work on an x86 or 64 linux distribution out of the box, is getting OpenOCD compiled on ARM that I think I would need to get working.

If I run https://github.com/arduino/OpenOCD-build-script on a raspberry pi, would the output work for the NRF51?
What is the difference in https://github.com/sandeepmistry/openocd-code-nrf5/ ? I assume this repository is what was used to create the executables listed in https://sandeepmistry.github.io/arduino-nRF5/package_nRF5_boards_index.json ?

sandeepmistry, 2017-01-03:
@carlosperate 

the following patches:

* https://github.com/sandeepmistry/openocd-code-nrf5/commit/7af8e2afea59ed9fd37b7689ac08250286d26dd3
* https://github.com/sandeepmistry/openocd-code-nrf5/commit/26ca830915275015be243a3fcc074856621e6d08