sandeepmistry/arduino-nRF5 #68: NRF 52 DK Error: No Valid JTAG Interface Configured.

womei, 2016-12-08:
Hello,

Even after replacing the drivers for the jtag, I'm still getting the following error:
`Local\Arduino15\packages\sandeepmistry\tools\openocd\0.10.0-dev.nrf5/bin/openocd.exe -d2 -f interface/jlink.cfg -c transport select swd; -f target/nrf52.cfg -c program {{C:\Users\XXX\AppData\Local\Temp\buildd1c2370d04f65e030b17edf3accf9382.tmp/sketch_dec07a.ino.hex}} verify reset; shutdown; 

Open On-Chip Debugger 0.9.0 (2015-05-19-12:06)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
Info : JLink SWD mode enabled
swd
adapter speed: 4000 kHz
cortex_m reset_config sysresetreq
Error: Cannot find jlink Interface! Please check connection and permissions.
Error: No Valid JTAG Interface Configured.
`

despite being able to talk to the board using the NRF GO application, help!

dlabun, 2016-12-08:
Do you know which firmware version of the onboard JLink using? There is a sticker on the board if you are unsure.

womei, 2016-12-09:
I beleive it was the outdated version first,
then Version 0219 (from https://developer.mbed.org/platforms/Nordic-nRF52-DK/)

and then a later nRFgo studio offered to install another firmware (might have been the JTAG or the nrf itself, not sure).

I will try flashing 0219 again and then uploading

dlabun, 2016-12-09:
That is not the correct firmware... You need to grab the JLink version and not the mbed version off the Nordic website:
[http://www.nordicsemi.com/eng/Products/Bluetooth-low-energy/nRF52-DK](url)

sandeepmistry, 2016-12-18:
Thanks @dlabun!

@womei please try out @dlabun's suggestion above and let us know.

closed 2016-12-18 by sandeepmistry

womei, 2017-04-05:
Apologies for the long delay, I recently got my DK back, I flashed the suggested J link firmware and re-replaced the drivers, but I am still receiving an issue:

Open On-Chip Debugger 0.10.0-dev-00254-g696fc0a (2016-04-10-10:13)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
swd
adapter speed: 10000 kHz
cortex_m reset_config sysresetreq
jaylink: Failed to open device: LIBUSB_ERROR_NOT_SUPPORTED.
Info : No device selected, using first device.
Info : J-Link OB-SAM3U128-V2-NordicSemi compiled Mar  2 2017 12:22:13
Info : Hardware version: 1.00
Info : VTarget = 3.300 V
Info : Reduced speed from 10000 kHz to 1000 kHz (maximum).
Info : Reduced speed from 10000 kHz to 1000 kHz (maximum).
Info : clock speed 10000 kHz
Info : SWD IDCODE 0x2ba01477
Info : nrf52.cpu: hardware has 6 breakpoints, 4 watchpoints
nrf52.cpu: target state: halted
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000548 msp: 0x20010000
** Programming Started **
auto erase enabled
Info : nRF51822-QFN48(build code: B00) 512kB Flash
Warn : using fast async flash loader. This is currently supported
Warn : only with ST-Link and CMSIS-DAP. If you have issues, add
Warn : "set WORKAREASIZE 0" before sourcing nrf51.cfg to disable it
wrote 4096 bytes from file C:\Users\XXX\AppData\Local\Temp\arduino_build_524776/Blink.ino.hex in 0.188412s (21.230 KiB/s)
** Programming Finished **
** Verify Started **
verified 2620 bytes in 0.048745s (52.489 KiB/s)
** Verified OK **
** Resetting Target **
shutdown command invoked

sandeepmistry, 2017-04-10:
```
** Programming Started **
auto erase enabled
Info : nRF51822-QFN48(build code: B00) 512kB Flash
Warn : using fast async flash loader. This is currently supported
Warn : only with ST-Link and CMSIS-DAP. If you have issues, add
Warn : "set WORKAREASIZE 0" before sourcing nrf51.cfg to disable it
wrote 4096 bytes from file C:\Users\XXX\AppData\Local\Temp\arduino_build_524776/Blink.ino.hex in 0.188412s (21.230 KiB/s)
** Programming Finished **
** Verify Started **
verified 2620 bytes in 0.048745s (52.489 KiB/s)
** Verified OK **
** Resetting Target **
shutdown command invoked
```

This looks good actually!

womei, 2017-04-10:
Ahhh, it was a combination of the 
`jaylink: Failed to open device: LIBUSB_ERROR_NOT_SUPPORTED.
Info : No device selected, using first device.`
bit and the fact I was trying to blink the LED on P0.17 which is connected to Arduino pin 6 😄 

it's working now, thanks so much!

salmanfarisvp, 2020-05-11:
@womei How did you solved the issues?