sandeepmistry/arduino-nRF5 #69: Error compiling in Windows 10

hulkco, 2016-12-09:
Went i try to compile, this its the error:
****
In file included from C:\Users\GReynaga\Documents\Arduino\libraries\BLEPeripheral\src/BLEPeripheral.h:24:0,

                 from C:\Users\GReynaga\Documents\Arduino\libraries\BLEPeripheral\examples\led\led.ino:6:

C:\Users\GReynaga\Documents\Arduino\libraries\BLEPeripheral\src/nRF51822.h:11:25: fatal error: ble_gatts.h: No such file or directory
***

How to correct the linux style path?

carlosperate, 2016-12-10:
Windows shouldn't care about the type of slash, are you sure the file exists in your pc?

sandeepmistry, 2016-12-18:
@hulkco please read: https://github.com/sandeepmistry/arduino-nRF5#ble

> All the relevant Nordic SoftDevice (S110, S130, S132) header files are included build path when a SoftDevice is selected via the Tools menu.

If this does not help, please enable verbose build logs in the Arduino IDE and provide full output.

closed 2016-12-18 by sandeepmistry

hulkco, 2016-12-18:
Thanks Dude!!!

tcpipchip, 2018-05-23:
Hi Sandee

I got today the NINA B112 (use NRF5 nordic), are you interested that i test the arduino on it ? I would like to offer it late for the clients. How is going the project ? Can users already implement beacons ? Hearbeat, etc...?