sandeepmistry/arduino-nRF5 #70: Support for NRF52840 Preview DK

lpercifield, 2016-12-18:
Awesome work on this core! I just received a couple of the new NRF52840 Dev Kits and hacked together support for it. It's still pretty dodgy, and is missing stuff.

Things that I had to do:
Add support for GPIO Port 1
Map GPIO pins
Update SDK files
Updated Softdevice files

https://github.com/lpercifield/arduino-nRF5/tree/nrf52840

Please let me know if you think I'm on the right track, happy for suggestions

Testing simple GPIO, Serial, and BLE with the BLEPeripheral Library, ALL WORKING

sandeepmistry, 2016-12-18:
@lpercifield awesome work!

You're fork seems to have lost a few changes from the master repo, but apart from that things look really good so far.

Regarding the GPIO port map stuff, maybe we should consider adding it to all of the existing variants?

I think we need new linker scripts because the nRF52840 has more RAM and flash.

lpercifield, 2016-12-18:
Thanks! Yeah I was in a hurry to get this code up, for backup purposes, so I didn't really pay attention to what was missing. I also noticed that there was a compiler error on the softdevice that I committed. Looks like the o's didn't get rebuilt and it was using the master version of the softdevice. I'll work on that, and update soon.

dlabun, 2016-12-18:
I just saw your comment on the compiler error on the soft device... They're already built from Nordic and shouldn't require anything on your part other than flashing them to the board

lpercifield, 2016-12-18:
It has to do with the softdevice API. I updated the s132 folder with what ships with the newest SDK and it doesn't compile properly as a drop in replacement. I haven't spent any time trying to fix the issue. I have a stupid git glitch with the original branch I created so I created a new one for the time being. I'll sort them both out when I have more time: https://github.com/lpercifield/arduino-nRF5/tree/nrf52840dk

dlabun, 2016-12-18:
Ah, got it. I will warn you there was a significant change in the Nordic SDK between the time this core was built and whats shipping today... Last I knew the current SDK (v12) was "broken" due to uppercase / lowercase naming issues.

lpercifield, 2016-12-18:
Thanks for the heads up! I'm sure that's what I ran into, as I was trying
code from SDK v13. Hope to have some time over the next couple of weeks to
bring it up to date. I'm really interested in getting to the s140
softdevice to try throughput of BLE 5, but first things first.

Are there any other cores related to NRF52 that are under active
development? Just want to ensure I'm not reinventing the wheel here.

honnet, 2017-03-31:
Hi @lpercifield, @dlabun and @sandeepmistry !
This nRF52840 developpement is very interesting, is there a TODO list somewhere ?
It would be great to have a USB-MIDI nRF52 RX module, would you have suggestions on how to start implementing it ?
Thanks again for this great work !
Cedric :)

sandeepmistry, 2017-04-17:
@honnet best way is to probably look at @lpercifield's repo https://github.com/lpercifield/arduino-nRF5/tree/nrf52840

honnet, 2017-04-17:
Perfect, thanks !

zipper555, 2018-03-06:
This is so soothing to hear when I'm struggling to port Arduino compatible libraries to nRF52840PDK.
So, is NRF52840 PDK usable with Arduino ide now?

knopserl, 2018-09-23:
@sandeepmistry and @lpercifield there where some start to support the nRF52840 with it's very very interesting BLE 5.0 (long range) options and ZigBee support. That is a big improvements to all other nRF5x MCU's. I'm /really looking forward for your great work. However there seem to be not very much progress since then and nRF52840 introduces a new SDK 15.2 for that BLE5.0, Thread, ZigBee stack which is based on the S140 softdevice. Is there any chance to get this supported in near future or at all?
To my understanding Aduino Studio and also Platform.io is relying and this project.

lpercifield, 2018-09-24:
@knopserl unfortunately, my work has taken me away from this project. I still love the Nordic MCUs and would love to get back into working on this branch, but the amount of time I have to contribute has been significantly reduced.

best of luck, and hopefully I can get back to it at some point

hanyazou, 2018-11-23:
I found that Adafruit had added nRF52840 and pca10056 support in their fork https://github.com/adafruit/Adafruit_nRF52_Arduino.

knopserl, 2018-11-25:
@hanyazou it's a good start. There are some popular nRF52840 missing such as the PCA10059 (USB dongle) and the modules from holyiot and EBYTE. That would be perfect.
Another promising and solder friendly board is this one from RF-Star:
https://www.alibaba.com/product-detail/2-4GHz-FCC-Nordic-nRF52840-Bluetooth_60816297852.html?spm=a2700.7724857.normalList.25.3b8011c1lf1Pc3

zdclaudy, 2019-05-06:
Any further movement on this?

mrdc, 2019-11-27:
> 
> 
> Any further movement on this?

It works fine with Generic nRF52 board.