sandeepmistry/arduino-nRF5 #71: Tools are slow to install

dlabun, 2016-12-24:
This conversation was started at the end of #57 regarding the slowness of the core's tools to install.

dlabun, 2016-12-24:
@sandeepmistry I ran a test with a modified fork, Using a Zip version of the tools will unpack in about 30 seconds while the existing bzip2 takes 3 minutes (This is with dual Xeons at 3.4Ghz so results will vary). Is Mac OS X capable of dealing with Zip files for the tools?

sandeepmistry, 2017-01-03:
@dlabun interesting!

> Is Mac OS X capable of dealing with Zip files for the tools?

Yes, the Arduino IDE uses the same Java code to extract archives.

The slowness might also be related to the compression level I selected.

In any case, do you have a size comparison for the files? Could you also prepare a PR for the JSON file and share links to the zip files you made?

dlabun, 2017-01-03:
@sandeepmistry On Windows the ZIP is ~120MB while the bzip2 is ~101MB. The download will be slightly bigger but the installation performance is significantly better. 

ARM only offers the Windows version as a ZIP so the Linux and Mac versions need to be repackaged first. How do you feel about pushing another tools release so the ZIPs accessible for testing?

sandeepmistry, 2017-01-16:
> How do you feel about pushing another tools release so the ZIPs accessible for testing?

@dlabun yes, let's go ahead. Do you have time to prepare the zip files and upload to the [`tools` release section](https://github.com/sandeepmistry/arduino-nRF5/releases/tag/tools)? I can test on OS X.

dlabun, 2017-01-16:
I can have the new ZIPs ready pretty fast... Do you want me to edit the existing tools release or start a tools v2 release?

sandeepmistry, 2017-01-16:
Let's start with just editing the existing tools (basically v0.2.0, but with zips for arm-gcc). Just to avoid changing too many things at once.

If this goes well we can make a v0.3.0 release afterwards.

dlabun, 2017-01-16:
ZIPs are up along with a text file with the check sums...  The files structures should be identical to the original versions.

sandeepmistry, 2017-01-16:
@dlabun nice! I see the zips there!

I think we should make a staging JSON (`package_nRF5_boards_index-staging.json`) in the `docs` folder for testing, then we can copy it over to the current one when ready. I'll try to do this tomorrow, but if you have time to work on it before I start let me know.

dlabun, 2017-01-16:
I may be able to get to it tonight, I'll let you know if I am able.

sandeepmistry, 2017-01-17:
I'm starting on this now :)

sandeepmistry, 2017-01-17:
@dlabun we're going to need to redo the zips unfortunately

```
fork/exec /Users/smistry/Library/Arduino15/packages/sandeepmistry/tools/gcc-arm-none-eabi/5_2-2015q4/bin/arm-none-eabi-g++: no such file or directory
```

```
/Users/smistry/Library/Arduino15/packages/sandeepmistry/tools/gcc-arm-none-eabi/5_2-2015q4/gcc-arm-none-eabi-5_2-2015q4-20151219-mac/gcc-arm-none-eabi-5_2-2015q4/bin/arm-none-eabi-g++
```

Is the path when I install with my JSON changes to use the zip, there are extra folder level `gcc-arm-none-eabi-5_2-2015q4-20151219-mac`, so the zips need this level removed.

dlabun, 2017-01-17:
Okay, I'll correct the ZIPs and repost once they're ready.

dlabun, 2017-01-17:
Were all of the ZIPs facing that issue? From looking at the ZIPs it's inconsistent on how many extra levels there may be.

sandeepmistry, 2017-01-30:
@dlabun I'll try to take another look at this in the next hour or 2 today.

sandeepmistry, 2017-01-30:
@dlabun sorry, I'll need to do this later this week, I underestimated my Github issue/PR backlog today.

dlabun, 2017-01-31:
@sandeepmistry No problem, when you get a chance to look into just let me know how you want the folder hierarchy to look

sandeepmistry, 2017-01-31:
So the zip format needs to be like: https://github.com/sandeepmistry/arduino-nRF5/releases/download/tools/gcc-arm-none-eabi-5_2-2015q4-20151219-mac.zip

The zip should contain a single root folder named `gcc-arm-none-eabi-5_2-2015q4`, that contains:
```
$ ls
arm-none-eabi	bin		lib		share
```

However, that one has issues and needs to be redone as some files in the `bin` folder and others lost executable permissions.

dlabun, 2017-01-31:
Okay I'll redo the ZIPs using that information. I'll look into the permissions issue, I am wondering if that occurs due to Zipping the files on Windows vs Mac.

sandeepmistry, 2017-02-02:
@dlabun thanks! I can test on Mac and we can get Travis CI to test out Linux.

dlabun, 2017-02-02:
@sandeepmistry new Zips are up. I used the ZIP from ARM for Windows, the Linux and Mac zips I created on my Mac mini... I think it may have been crying from trying to compress those files.

sandeepmistry, 2017-02-02:
@dlabun cool, it's weird I'm still getting:

```
"/Users/smistry/Library/Arduino15/packages/sandeepmistry/tools/gcc-arm-none-eabi/5_2-2015q4/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mthumb -c -g -Os -w -mfloat-abi=hard -mfpu=fpv4-sp-d16 -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions  -w -x c++ -E -CC -DF_CPU=16000000 -DARDUINO=10612 -DARDUINO_NRF52_DK -DARDUINO_ARCH_NRF5  -DNRF52 -DNRF5 "-I/Users/smistry/Library/Arduino15/packages/sandeepmistry/hardware/nRF5/0.2.0/cores/nRF5/SDK/components/toolchain/" "-I/Users/smistry/Library/Arduino15/packages/sandeepmistry/hardware/nRF5/0.2.0/cores/nRF5/SDK/components/toolchain/CMSIS/Include" "-I/Users/smistry/Library/Arduino15/packages/sandeepmistry/hardware/nRF5/0.2.0/cores/nRF5/SDK/components/toolchain/gcc/" "-I/Users/smistry/Library/Arduino15/packages/sandeepmistry/hardware/nRF5/0.2.0/cores/nRF5/SDK/components/device/" "-I/Users/smistry/Library/Arduino15/packages/sandeepmistry/hardware/nRF5/0.2.0/cores/nRF5/SDK/components/drivers_nrf/delay/" "-I/Users/smistry/Library/Arduino15/packages/sandeepmistry/hardware/nRF5/0.2.0/cores/nRF5/SDK/components/softdevice/none/headers/" -DUSE_LFXO "-I/Users/smistry/Library/Arduino15/packages/sandeepmistry/hardware/nRF5/0.2.0/cores/nRF5" "-I/Users/smistry/Library/Arduino15/packages/sandeepmistry/hardware/nRF5/0.2.0/variants/nRF52DK" "/var/folders/5d/pmvtw7p579n5npm6p421qf540000gn/T/arduino_build_71976/sketch/Blink.ino.cpp" -o "/var/folders/5d/pmvtw7p579n5npm6p421qf540000gn/T/arduino_build_71976/preproc/ctags_target_for_gcc_minus_e.cpp"
fork/exec /Users/smistry/Library/Arduino15/packages/sandeepmistry/tools/gcc-arm-none-eabi/5_2-2015q4/bin/arm-none-eabi-g++: permission denied
Error compiling for board nRF52 DK.
```

```
$ ls -l /Users/smistry/Library/Arduino15/packages/sandeepmistry/tools/gcc-arm-none-eabi/5_2-2015q4/bin/
total 64552
-rw-r--r--  1 smistry  staff   898968 19 Dec  2015 arm-none-eabi-addr2line
-rw-r--r--  1 smistry  staff   933192 19 Dec  2015 arm-none-eabi-ar
-rw-r--r--  1 smistry  staff  1596324 19 Dec  2015 arm-none-eabi-as
-rw-r--r--  1 smistry  staff   960288 19 Dec  2015 arm-none-eabi-c++
-rw-r--r--  1 smistry  staff   898500 19 Dec  2015 arm-none-eabi-c++filt
-rw-r--r--  1 smistry  staff   960208 19 Dec  2015 arm-none-eabi-cpp
-rw-r--r--  1 smistry  staff    38612 19 Dec  2015 arm-none-eabi-elfedit
-rw-r--r--  1 smistry  staff   960288 19 Dec  2015 arm-none-eabi-g++
-rw-r--r--  1 smistry  staff   960120 19 Dec  2015 arm-none-eabi-gcc
-rw-r--r--  1 smistry  staff   960120 19 Dec  2015 arm-none-eabi-gcc-5.2.1
-rw-r--r--  1 smistry  staff    44056 19 Dec  2015 arm-none-eabi-gcc-ar
-rw-r--r--  1 smistry  staff    44184 19 Dec  2015 arm-none-eabi-gcc-nm
-rw-r--r--  1 smistry  staff    44192 19 Dec  2015 arm-none-eabi-gcc-ranlib
-rw-r--r--  1 smistry  staff   517960 19 Dec  2015 arm-none-eabi-gcov
-rw-r--r--  1 smistry  staff   479388 19 Dec  2015 arm-none-eabi-gcov-tool
-rw-r--r--  1 smistry  staff  5573708 19 Dec  2015 arm-none-eabi-gdb
-rw-r--r--  1 smistry  staff  5821076 19 Dec  2015 arm-none-eabi-gdb-py
-rw-r--r--  1 smistry  staff   974100 19 Dec  2015 arm-none-eabi-gprof
-rw-r--r--  1 smistry  staff  1270948 19 Dec  2015 arm-none-eabi-ld
-rw-r--r--  1 smistry  staff  1270948 19 Dec  2015 arm-none-eabi-ld.bfd
-rw-r--r--  1 smistry  staff   913056 19 Dec  2015 arm-none-eabi-nm
-rw-r--r--  1 smistry  staff  1110512 19 Dec  2015 arm-none-eabi-objcopy
-rw-r--r--  1 smistry  staff  1400632 19 Dec  2015 arm-none-eabi-objdump
-rw-r--r--  1 smistry  staff   933192 19 Dec  2015 arm-none-eabi-ranlib
-rw-r--r--  1 smistry  staff   517608 19 Dec  2015 arm-none-eabi-readelf
-rw-r--r--  1 smistry  staff   903500 19 Dec  2015 arm-none-eabi-size
-rw-r--r--  1 smistry  staff   899016 19 Dec  2015 arm-none-eabi-strings
-rw-r--r--  1 smistry  staff  1110512 19 Dec  2015 arm-none-eabi-strip
```

Let me try to delete the packages folder and try again. Previously I had uninstalled the core via the IDE's board manager.

sandeepmistry, 2017-02-02:
Same thing again :(

Looking at ESP8266 JSON, they still use .tar.gz's http://arduino.esp8266.com/versions/2.3.0/package_esp8266com_index.json

Another idea is to remove extra stuff from the archives (like docs in `share/doc/gcc-arm-none-eabi/`). I think that's what the SAMD cores GCC package does ...

sandeepmistry, 2017-02-02:
@cmaglie does Arduino use a script to repackage `arm-none-eabi-gcc/4.8.3-2014q1` from ARM?

dlabun, 2017-02-02:
@sandeepmistry That's just really strange... Didn't the Board Manager code change recently?

dlabun, 2017-02-02:
@sandeepmistry Or maybe we're looking at this wrong, Maybe this is an issue with files coming from ARM.

sandeepmistry, 2017-02-06:
@dlabun nothing in the boards manager has changed as far as I can tell. However, from my understanding zip files don't preserve read, write, and executable file attributes, so unless the extractor code in the IDE forces executable permissions on all extracted files I wouldn't expect it to work.

Thoughts?

dlabun, 2017-02-06:
@sandeepmistry I'm not 100% about zips and file permissions, but here's the weird thing... If I unzip the files on my Mac using the macOS built in tools I can copy / paste them over to Arduino and they don't have any permission issues.

How are testing the ZIP files? I couldn't get the staging JSON to work for some reason, the IDE downloads it but it won't show up in the board manager.

sandeepmistry, 2017-02-06:
How are you zipping them? I was testing with the command line.

I edited the JSON file locally, then served it using the the `http-server` module from npm for Node.js

Prior this I had removed the original core installation, and delete the nRF5 files from `~/Library/Arduino15/packages/sandeepmistry` and ` ~/Library/Arduino15/staging/packages`, then restarting the IDE.

```
{
  "packages": [
    {
      "name": "sandeepmistry",
      "maintainer": "Sandeep Mistry",
      "websiteURL": "https://github.com/sandeepmistry/arduino-nRF5",
      "email": "sandeep.mistry@gmail.com",
      "help": {
        "online": "https://github.com/sandeepmistry/arduino-nRF5/issues"
      },
      "platforms": [
        {
          "name": "Nordic Semiconductor nRF5 Boards",
          "architecture": "nRF5",
          "version": "0.1.0",
          "category": "Contributed",
          "help": {
            "online": "https://github.com/sandeepmistry/arduino-nRF5/issues"
          },
          "url": "https://github.com/sandeepmistry/arduino-nRF5/archive/0.1.0.tar.gz",
          "archiveFileName": "arduino-nRF52-0.1.0.tar.gz",
          "checksum": "MD5:1825f6420104225f5fee11b467923b7d",
          "size": "591297",
          "boards": [
            {"name": "nRF52 DK"},
            {"name": "Bluz DK"}
          ],
          "toolsDependencies": [
            {
              "packager": "sandeepmistry",
              "name": "gcc-arm-none-eabi",
              "version": "5_2-2015q4"
            },
            {
              "packager": "sandeepmistry",
              "name": "openocd",
              "version": "0.10.0-dev.nrf5"
            }
          ]
        },
        {
          "name": "Nordic Semiconductor nRF5 Boards",
          "architecture": "nRF5",
          "version": "0.2.0",
          "category": "Contributed",
          "help": {
            "online": "https://github.com/sandeepmistry/arduino-nRF5/issues"
          },
          "url": "https://github.com/sandeepmistry/arduino-nRF5/archive/0.2.0.tar.gz",
          "archiveFileName": "arduino-nRF52-0.2.0.tar.gz",
          "checksum": "MD5:1ed95ee9030e4ca1371665258a3cdee4",
          "size": "562847",
          "boards": [
            {"name": "nRF52 DK"},
            {"name": "Bluz DK"},
            {"name": "RedBearLab BLE Nano"},
            {"name": "RedBearLab nRF51822"},
            {"name": "BBC micro:bit"},
            {"name": "OSHChip"},
            {"name": "Generic nRF51822"}
          ],
          "toolsDependencies": [
            {
[... 62 lines cut ...]
              "size": "3745903",
              "checksum": "MD5:045026b97051debf2e78838ea5623e5d"
            },
            {
              "host": "x86_64-pc-linux-gnu",
              "url": "https://github.com/sandeepmistry/arduino-nRF5/releases/download/tools/openocd-linux32-0.10.0-dev-nrf5.zip",
              "archiveFileName": "openocd-linux32-0.10.0-dev-nrf5.zip",
              "size": "3745903",
              "checksum": "MD5:045026b97051debf2e78838ea5623e5d"
            },
            {
              "host": "i686-mingw32",
              "url": "https://github.com/sandeepmistry/arduino-nRF5/releases/download/tools/openocd-win32-0.10.0-dev-nrf5.zip",
              "archiveFileName": "openocd-win32-0.10.0-dev-nrf5.zip",
              "size": "5498373",
              "checksum": "MD5:1e03019a21206bf79750f946a3b7030c"
            }
          ]
        }
      ]
    }
  ]
}

```

dlabun, 2017-02-06:
I just doubled clicked the ZIPs and the Mac automatically unzipped them... I'm that good with the Mac command line. 

I'll try them through the Board Manager like you did just to see how they behave on my machine.

sandeepmistry, 2017-02-07:
Ok, I think we'll have a similar issue with Linux, but let's verify Mac behaviour first ...

dlabun, 2017-02-07:
Looks like there's an issue with Windows as well... 

[image]

sandeepmistry, 2017-02-07:
Right, we'll have to repackage it with a root folder like https://github.com/sandeepmistry/arduino-nRF5/issues/71#issuecomment-276261755

dlabun, 2017-02-07:
That was repackaged... The unzipped version doesn't look anything like the zipped.

The upper set of folders is the root level of the zip, the lower is how it looks after the IDE unzips it

[image]

sandeepmistry, 2017-02-10:
@dlabun what's the verdict on this?

Here's some potential ways forward:
 * Switch to zips for Windows only
 * Try to remove some of the files like documentation from the GCC tar file, to see if that improves installation times. 

Any other suggestions?

dlabun, 2017-02-10:
@sandeepmistry To be honest, it's a one time (or very infrequent) annoyance when you install or update the core... For the time being it might be best to leave the core as is, add a note in the readme and focus on fixing other issues. With that being said, I am going to keep looking into this when time allows as I think the issue is with the board manager itself.

sandeepmistry, 2017-08-20:
Closing this for now based on the last comment.

closed 2017-08-20 by sandeepmistry