sandeepmistry/arduino-nRF5 #74: PLease help with flashing

TraxXavier, 2017-01-05:
I got a few of these boards: http://aconno.de/wp-content/uploads/2016/10/ACN52832-Data-Sheet.pdf
and a ST-Link V2 programmer https://www.amazon.de/gp/product/B012VR3PVA/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1

I connected it like following
board   programmer
1   to   2
2   to   6
3   to   1
4   to   4
5   to   7
6   to   8

And tried to flash with the option nRF52 DK

the From git (for core development) does not work for me (i have arduino-1.6.12)
it does not show the new boards,
the boardmanager way works but does not show an entry for generic MCU

But it days that there is no j-tag link set up

I tried to use the STM32 ST-LINK Utility to connect but that also fail.
I used a LA to look at the pins and noticed that the state of the reset pin does not change is that normal? I selected the HW reset option in the utility, so i would expect the reset pin to do something.

please help me to get the things to run, 
thanks

dlabun, 2017-01-05:
That's not a real ST-LINK/V2... It's possible you picked up a badly made clone that's not working properly.

TraxXavier, 2017-01-05:
ok thats not good so i need to get an other.

Also how do i get a code version that actually shows the "Generic nRF52" entry?

cheers

dlabun, 2017-01-07:
It's a SWD interface and you need to follow the instructions given to you in your other issue ticket.

d00616, 2017-01-07:
@TraxXavier I use these adapters: http://www.ebay.de/itm/172287103596 but i think you have an error in your wiring. GND is connected to 3,3V

> Your connection:
> board programmer
> 1 (GND) to 2 (3,3V)
> 2 to 6 (SWDIO)
> 3 to 1 (RST)
> 4 to 4 (SWCLK)
> 5 (P0.28) to 7 (GND)
> 6 to 8 (JTDO)

Please check my suggestion. I have changed the Programmer and Board order!
Programmer -> Board
2 -> 12, 23, 24 or 32
4 -> 34
6 -> 33
7 -> 1, 13 or 35

A JTAG interface is not available.

TraxXavier, 2017-01-14:
I got a original programmer: http://uk.rs-online.com/web/p/entwicklungskits-prozessor-mikrocontroller/7141701/

> 2 -> 12, 23, 24 or 32
> 4 -> 34
> 6 -> 33
> 7 -> 1, 13 or 35

2 is SWCLK and it should go to 12, 23, 24 or 32 these are VCC on the board 
4 is SWDIO tp oin 34 on the borad and these should go to SWDCLK IC pin 25
6 GND to pin 33 to SWDIO, pin 26 on the chip
7 3.3 v to 1, 13 or 35 what is GND

Are you sure that is how its intended? cause it looks unusual to say the least.

Anyhow I tried to connect using the new programmer, but still no success :/

Also how do I make the arduino IDE to stip trying JTAG?
And as mentioned already I don't have the option for a generic chip in the board list how can I fix that?

d00616, 2017-01-14:
@TraxXavier My suggestion of cabling is for the "WINGONEER ST-Link Stlink v2". Please refer to user manual chapter 6.2 for cabling.

> 2 is SWCLK and it should go to 12, 23, 24 or 32 these are VCC on the board
2 is 3.3V at programmer side and not SWCLK
> 4 is SWDIO tp oin 34 on the borad and these should go to SWDCLK IC pin 25
4 is SWCLK at programmer side and not SWDIO
> 6 GND to pin 33 to SWDIO, pin 26 on the chip
6 is SWDIO at programmer side and not GND
> 7 3.3 v to 1, 13 or 35 what is GND
7 is GND at programmer side and not 3.3V

TraxXavier, 2017-01-14:
ok, i see... folowing that once datasheet 
2 vcc
7 gnd
4 SWCLK
6 SWDIO

My wiring with the original ST-Link V2 was similar i also had rst and SWO connected
Should I just try removing those two wires?

Also how do I make the arduino IDE to stip trying JTAG?
And as mentioned already I don't have the option for a generic chip in the board list how can I fix that?

d00616, 2017-01-14:
> Should I just try removing those two wires?

I don't know. I use only SWCLK and SWDIO.

> Also how do I make the arduino IDE to stip trying JTAG?

What is "stip"? How do you want to use JTAG with a MCU which doesn't support it?

> And as mentioned already I don't have the option for a generic chip in the board list how can I fix that?

The is an Board "Generic nRF52".

TraxXavier, 2017-01-14:
> What is "stip"? How do you want to use JTAG with a MCU which doesn't support it?
I meant to stop,

When I try to use the arduino IDE with teh sources from this repo it always gives me an error about jlink and jtag.

So I guess I somehow have to tell it ti not use JTAG but SWD

What board option are you using? In my arduino IDE I cant find a generic nRF52 only the DK

d00616, 2017-01-14:
> When I try to use the arduino IDE with teh sources from this repo it always gives me an error about jlink and jtag.

J-Link is another programmer. You have to choose "ST-Link v2" from "Programmer" menu.

d00616, 2017-01-14:
> What board option are you using? In my arduino IDE I cant find a generic nRF52 only the DK

The Generic nRF52 is available via git and not released at the moment.

TraxXavier, 2017-01-14:
I have selected the "ST-Link v2" programmer yet I still get the following error when trying to upload a sketch:
"Open On-Chip Debugger 0.10.0-dev-00254-g696fc0a (2016-04-10-10:13)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 0
swd
adapter speed: 10000 kHz
cortex_m reset_config sysresetreq
Error: No J-Link device found.
Error: No Valid JTAG Interface Configured.
Error: No Valid JTAG Interface Configured.
"

Also how do I get the git version to work with my arduino ide to get the support for a Generic nRF52

d00616, 2017-01-14:
@TraxXavier Remove the board in Arduino and follow https://github.com/sandeepmistry/arduino-nRF5#from-git-for-core-development

TraxXavier, 2017-01-14:
Ok, now i have the generic option, but compile fails with the error:
exec: "/bin/arm-none-eabi-g++": file does not exist
Error compiling for board Generic nRF52.

I'm using arduino ide 1.6.12 under windows

dlabun, 2017-01-14:
You'll get that error if you do not follow the installation instructions. You need to follow the instructions listed under Board Manager, followed by Flashing a Soft Device, and lastly From git (for core development).

TraxXavier, 2017-01-14:
ok so i shouldn't have removed the instaled board i see...

now i have the generic nRF52,

howeever the menu option "nRF5 Flash SoftDevice" fails

Open On-Chip Debugger 0.10.0-dev-00254-g696fc0a (2016-04-10-10:13)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 0
0x4000
adapter speed: 10000 kHz
Error: open failed
in procedure 'init' 
in procedure 'ocd_bouncer'

Error while flashing SoftDevice.

The LED on the genuine ST Link V2 stays red all the time

I followed this pinout: http://embedded-lab.com/blog/wp-content/uploads/2014/08/397903258_308.jpg for SWD and the one on site 14 of http://aconno.de/wp-content/uploads/2016/10/ACN52832-Data-Sheet.pdf

I supply the board with 3.3V externally,
connected 
VCC (J1 Pin 1 of the board's programming port) to pin 1 on the ST Link V2
SWDIO (J1 Pin 2, chip pin 26) to pin 7
SWDCLK(J1 Pin 4, chip pin 25) to pin 9
GND (J1 Pin 5) to GND

nRESET (J1 Pin 3, chip pin 24) to pin 15
SWO(J1 Pin 6, chip pin 21) to pin 13

so what should I try next?

dlabun, 2017-01-14:
Did you install the ST Link drivers? It doesn't look like OpenOCD is even connecting to the STLink.

TraxXavier, 2017-01-14:
Yes it should be installed its detected as "STMicroelectronics STLink dongle"
and their tool "STM32 ST-LINK Utility" detects it,
how ever also it can not connect to the chip :/

dlabun, 2017-01-14:
Did you select the correct COM port in the IDE?

TraxXavier, 2017-01-14:
what com port? the programmer does not emulate a comport

nonsintetic, 2017-01-14:
Here's a checklist, I know all these were posted, but maybe having it all together will help:

- if you use an ST Link (i use the cheapo chinese ones, but should be same) all you need to connect is:
  + SWDIO (it's pin 7 on your programmer's SWD connector and pin 33 on your nrf module)
  + SWDCLK (pin 9 on your programmer's SWD connector and pin 34 on your module)
  + GND (pin 4 on programmer and pin 1 on module)
  + VCC (pin 1 on programmer and pin 12 on module)
- check the connections with a multimeter in case of bad solder joints. If you have an oscilloscope you can probe the SWDIO and SWDCLK while programming to see if they actually do anything.
- check if you installed the java file for soft device flashing in the tools folder correctly, maybe do it again just to be safe
- pretty sure you don't need to flash a soft device to get it to run a sketch that doesn't use bluetooth so you can try a simple blink or something to see if it works. If you want to flash a soft device make sure you select the version in the Tools > Softdevice menu
- select ST-Link v2 from Tools > Programmer
- make sure you have a board package installed in the Arduino IDE that has the ARM compiler tool-chain (you can install the Arduino SAMD Boards package for example). Just having it installed is enough for arduino to use it on any ARM board.
- try to upload a blink sketch. your board has an RGB led connected to the pins: 22, 23, 24 (one for each color) so you can use those in your sketch.

If it doesn't work after all that you could try and get a 3 dollar cheapo ST Link, since that's what most people will have (search for 'Mini ST-Link V2' on E-bay). It's not really money wasted, you can flash the Black Magic Probe firmware on it if it doesn't work and it'll be a great tool to have for programming all sorts of stuff.

PS: if you have that 'Tag connect' thing they mention in the datasheet of the module, that's a J-Link, I think you can use that to program it instead of the ST Link (not 100% sure).

dlabun, 2017-01-14:
@nonsintetic You're incorrect about Tag Connect... That is a brand name connector used for programming.

dlabun, 2017-01-15:
@TraxXavier Ignore the COM port comment, I was thinking of another board. The LED on the STLink will flash with activity... If you don't see any blinking when you attempt to program it's a sure sign there's a communication issue between OpenOCD and the STLink.

d00616, 2017-01-15:
@TraxXavier like explained [here](https://github.com/sandeepmistry/arduino-nRF5/issues/74#issuecomment-271097935), [here](https://github.com/sandeepmistry/arduino-nRF5/issues/74#issuecomment-272625241) and [here](https://github.com/sandeepmistry/arduino-nRF5/issues/74#issuecomment-272661975), your cabling of your nRF52 board looks incorrect.

You have to connect only these board pins correctly to your programmer:
[image]

This is the screenshot of the first rows of the pin assignments table:
[image]

TraxXavier, 2017-01-15:
I connected the programmer like shown in d00616's image but i still cant get a connection using the "STM32 ST-LINK Utility" the led on the programmer blinks and then i get an error from the app.

when I try to use the arduino IDE the LED of the programmer does not do anything

So it seams I still have two issues one is connection between the arduino IDE and the programmer and the other is the programmer itself hmpf....

how does your programmer ST-Link V2 whos in the device manager, do i need the driver from the OEM or may be something 3rd party?

dlabun, 2017-01-15:
@TraxXavier You will never get it to work using the STM32 utility, that program is only intended to talk to STM32 chips.

dlabun, 2017-01-15:
@d00616 Is the reset line required? I'm mostly working with the DKs so I haven't looked at the SWD wiring closely.

TraxXavier, 2017-01-15:
ok so good to know that I can ignore the STM32 utility issue, stil...
the communication issue remains why the ardhino IDE does not talk to my ST-Link V2

what drivers do you use the original once from STM or some 3rd party once?

dlabun, 2017-01-15:
I honestly don't know about the driver, I don't use the STLink personally. I know the Segger J-Link does require a 3rd party driver, it's entirely possible the STLink needs it as well.

TraxXavier, 2017-01-15:
I found the issue, for some reason Open OCD does not work on a USB 3 port, using a USB 2 port on my PC I get a connection.

WTF WTFFFF!!!

what a stupid issue I cant believe it!!!

Now I'm going to band my had against the nearest wall really hard....

Thanks for all the help :)

dlabun, 2017-01-15:
Are you able to communicate with the module now?

TraxXavier, 2017-01-15:
yes, i could upload a blink sketch for the red led on pin 22
I'll try tomorrow something with Bluetooth :)

dlabun, 2017-01-15:
Closing as resolved.

closed 2017-01-15 by dlabun

sandeepmistry, 2017-01-16:
@dlabun thanks for helping out!

@TraxXavier could you please submit an update the read me appropriately to help others avoid the issue in the future?