sandeepmistry/arduino-nRF5 #76: Update Softdevice S132 to version 3.0.1 problems

nestorsilk, 2017-01-07:
Changed the path definition in nRF5FlashSoftDevice.java. Compiled into .jar. Then replaced the existing file within the tools folder. Also exchanged the softdevice .hex file within the SDK softdevice folder for S132 and all the other necessary files.

But when trying to flash the soft device to the board I get an openocd .tcl script error. Openocd can't find the softdevice .hex file for version 2.0.1.  Although it should already look for version 3.0.1. So, this seems to be hardcoded somewhere else. Couldn't find it. 

Would be thankful for any hint.

dlabun, 2017-01-07:
You need to update boards.txt with the appropriate soft device version.

nestorsilk, 2017-01-07:
Yes, forgot this! Everything works now.

Thanks a lot.

nestor

closed 2017-01-07 by nestorsilk

SL06, 2019-02-26:
Can we update to the latest version (6.1.1) from this method and does it work ?  Could you write a detailed procedure ?