sandeepmistry/arduino-nRF5 #78: Timer problems

nonsintetic, 2017-01-12:
Hello,

I've been trying to get some real time clock functionality going with the core and the time seems to always be running faster than normal regardless of hardware. The errors I'm getting are pretty big, 1-5 seconds extra every minute, so it couldn't be just the crystal's PPM error.

I've tried counting seconds based on millis() and also the Time library by PaulStoffregen (which does the same when not linked to a time source).  I've tested on the BLE Nano and the Waveshare Core51822 boards and it seems to be the same.

Any ideas what could be causing this? Could it be a software issue in the core or a hardware one?

Thank you for all the great work!

dlabun, 2017-01-12:
Could you post the code you are trying to use?

nonsintetic, 2017-01-12:
Here it is, I'm using a SHARP memory display on hardware SPI with it too.

```#include "SPI.h"
#include "Adafruit_GFX.h"
#include <Adafruit_SharpMem_SPI.h>

//display stuff
Adafruit_SharpMem display(SS);
#define BLACK 0
#define WHITE 1

// Use these variables to set the initial time
int hours = (__TIME__[0] -'0') *10 +( __TIME__[1] -'0') ;
int minutes = (__TIME__[3] -'0') *10 +( __TIME__[4] -'0') ;
int seconds = (__TIME__[7] -'0') *10 +( __TIME__[7] -'0') ;
unsigned long lastUpdate = 0;

#define DEBUG_RTC_DISPLAY 1
#define DEBUG_RTC_SERIAL 1

void setup() {
  Serial.begin(9600);
  Serial.println("RTC Test");
  display.begin();
  display.setTextSize(2);
  display.setTextColor(BLACK,WHITE);
  display.setCursor(0,0);
  display.clearDisplay();
  display.refresh();
}

void loop() {
  updateTime(); //update the RTC
}

void updateTime() {
   //Serial.print(lastUpdate);
   //Serial.print(" ");
   //Serial.println(millis());
    if(millis() - lastUpdate >= 1000) {
    lastUpdate = millis();
    seconds++;  // Increment seconds
    if (seconds >= 59)  // If seconds overflows (>=60)
    {
      seconds = 0;  // Set seconds back to 0
      minutes++;    // Increment minutes
      if (minutes >= 59)  // If minutes overflows (>=60)
      {
        minutes = 0;  // Set minutes back to 0
        hours++;      // Increment hours
        if (hours >= 23)  // If hours overflows (>=23)
        {
          hours = 0;  // Set hours back to 0
        }
      }
    }
    #ifdef DEBUG_RTC_SERIAL
      Serial.print(hours);
      Serial.print(":");
      Serial.print(minutes);
      Serial.print(":");
      Serial.println(seconds);
    #endif
    #ifdef DEBUG_RTC_DISPLAY
      display.setCursor(18,20);
      if(hours < 10) display.print(0);
      display.print(hours);
      display.print(":");
      if(minutes < 10) display.print(0);
      display.println(minutes);
      display.setCursor(18,50);
      if(seconds < 10) display.print(0);
      display.println(seconds);
      display.refresh();
    #endif
  }
}
```

dlabun, 2017-01-12:
I think your problem is the over head of the LCD library updating the display followed by the time it takes to actually process the time calculations. I have a very similar setup with a Sharp memory LCD but I poll an external real time clock for an accurate time.

nonsintetic, 2017-01-12:
I updated my reply with a simplified version of the test that omits the LCD and just outputs to serial. So that can't be it..

dlabun, 2017-01-12:
Unfortunately I only have an nRF52 demo board but I'll atleast test it and see how accurate it is.

dlabun, 2017-01-12:
You may want to give this a read while I am looking into this issue, it could address this issue:

[https://devzone.nordicsemi.com/question/53893/best-way-to-implement-system-clock-on-nrf51/?answer=54538#post-id-54538](url)

nonsintetic, 2017-01-12:
That uses app_timer, that's part of the Nordic SDK which I believe is not included in this repo. I think the millis() implementation in this core uses the hardware timer already, but the devil is in the details. I'll keep looking around to see if I can find something relevant.

d00616, 2017-01-15:
This updates seconds from millis(). Btw your code fails after 49 days and 17 hours by an overflow.

```
unsigned long seconds = 0 ;
unsigned long lastUpdate = 0;
unsigned long overflows = 0;
unsigned long lastPrint = 1;

void setup() {
  Serial.begin(9600);
  Serial.println("RTC Test");
}

void loop() {
  updateTime(); //update the RTC
  if (lastPrint != seconds) {
    Serial.print(seconds);
    Serial.print(" ");
    Serial.println(millis());
    lastPrint = seconds;
  }
}

void updateTime() {
    unsigned long mymillis = millis() / 1000;
    if(mymillis != lastUpdate) {
      lastUpdate = mymillis;
      seconds = mymillis + overflows;
      if (lastUpdate > mymillis) overflows+=(4294967296/1000);
  }
}
```

nonsintetic, 2017-01-15:
That's all fine and dandy, the code works fine and is better than mine, but **the clock is still faster than it should be**. I'm comparing it with UNIX time from the computer (the first column in my test is added by RealTerm, it's the PC's time so it should be as accurate as possible). Here's a test with the code you posted:

```1484481980,0
1484481981,1
1484481982,2
1484481983,3
1484481984,4
1484481985,5
1484481986,6
1484481986,7
1484481987,8
1484481988,9
1484481989,10
1484481990,11
1484481991,12
1484481992,13
1484481993,14
1484481994,15
1484481995,16
1484481996,17
1484481996,18
1484481997,19
1484481998,20
1484481999,21
1484482000,22
1484482001,23
1484482002,24
1484482003,25
1484482004,26
1484482005,27
1484482006,28
1484482007,29
1484482007,30
1484482008,31
1484482009,32
1484482010,33
1484482011,34
1484482012,35
1484482013,36
1484482014,37
1484482015,38
...```

d00616, 2017-01-15:
@nonsintetic Sorry for my missunderstanding. Both boards you use are haven't enabled the "Low Frequency Oscillator" -> -DUSE_LFXO

Both boards have an soldered LFXO, but it is not enabled in boards.txt.

dlabun, 2017-01-15:
If either of you would like to create a pull request to fix this it would be appreciated.

nonsintetic, 2017-01-15:
That could very well explain it, the RC is way less precise.
But I just uploaded the example with the 'Generic NRF51822' board selected to my Waveshare board, with the Low Frequency Clock set to 'Crystal Oscillator' (it's `Generic_nRF51822.menu.lfclk.lfxo.build.lfclk_flags=-DUSE_LFXO` in the boards file I have) and it seems to be the same.
Maybe the flag isn't working? How could I test if it's really using the external crystal, could I see it on the scope if I probe the pins or are they always on?

d00616, 2017-01-15:
> If either of you would like to create a pull request to fix this it would be appreciated.

@dlabun Sorry, i have no time to do this at the moment.

nihaopaul, 2017-01-15:
i normally test the build flags that it compiles with but yes agree, current is normally higher when using the internal vs external crystal also if you remove the external crystal and set it to use the external crystal you wont be able to send anything over bluetooth, if you can, it's using the internal one. if it has a HFCK then use it! power savings are great.

nonsintetic, 2017-01-15:
How would I go about using the HFCK? It does have one fitted on the board. Is that what the 'synthesized' option does?
In any case, if it's not using the external crystal, then there's a bug somewhere.

sandeepmistry, 2017-01-16:
The code in [cores/nRF5/wiring.c](https://github.com/sandeepmistry/arduino-nRF5/blob/master/cores/nRF5/wiring.c) is currently hardcoded for the RC:

```
  NRF_CLOCK->LFCLKSRC = (uint32_t)((CLOCK_LFCLKSRC_SRC_RC << CLOCK_LFCLKSRC_SRC_Pos) & CLOCK_LFCLKSRC_SRC_Msk);

```

The flags is only used in the BLEPeripheral library currently. Could something try changing the above line to select the appropriate LF CLK SRC based on the menu item/define? 

If it works well, please submit a pull request.

nonsintetic, 2017-01-16:
It works! I submitted a pull request #81 . 
With the Crystal Oscillator or Synthesized Clock selected the timing is perfect, no more 1 second drifts. I'll add the menu options for the other boards that have external oscillators too in a different pull request.

sandeepmistry, 2017-02-18:
Fixed by #96 (rebased version of #81).

closed 2017-02-18 by sandeepmistry