sandeepmistry/arduino-nRF5 #79: SPI doesn't work when using BLEPeripheral and Crystal Oscillator

shlomozippel, 2017-01-13:
Steps to reproduce:

1. Use RF51 board with crystal oscillator (I used nRF51 DK)
2. Use BLEPeripheral library. It is enough to call BLEPeripheral::begin() to init the softdevice
3. Use SPI library. SPI.begin(); SPI.transfer(0); is enough

The transfer function hangs - the while (!SPI->EVENTS_READY) loop never returns. This only happens when the soft device is initialized and only happens when clock source is set to crystal. If I change clock source to internal RC then SPI works as expected also with the softdevice initialized.

nonsintetic, 2017-01-13:
I've tested this on a **Waveshare Core51822** board using the '_Generic NRF51_' profile and a **SHARP memory display on SPI**. I flashed the S110 soft device using the tool provided, set the low frequency clock to crystal oscillator and just added a few lines to the 'serial' example to get the SPI screen to print the millis() each loop. 
**It all works fine.** I can connect to the board using the Bluefruit LE app, the screen shows the millis(), doesn't seem to hang at all. 
Maybe you can post some simple code for testing, maybe it's something in the code you're using or only specific to the DK.

PS: sometimes i get issues with the sketch not flashing if I'm using BLE and I don't flash the soft device before, maybe try flashing the soft device and then uploading right after.

dlabun, 2017-01-13:
Please post your code so we can take a look at this issue.

shlomozippel, 2017-01-13:
Oops.

I just double checked everything and this is happening because I am using pins 26 and 27 as SPI pins -- but they are already used for the oscillator. My board doesn't have an external oscillator so it only happened when I was simulating on the nrf51DK.

Thanks for looking into this!

closed 2017-01-13 by shlomozippel