sandeepmistry/arduino-nRF5 #80: Fixing Driver

boppana113, 2017-01-16:
I followed the instructions for using the Arduino IDE, but soon later I discovered NRFgo did not recognize my NRF52 DK. I think it had something to with using Zadig to replace the driver. Is there anyway to restore the driver back to its original settings so I can program it using keil.

dlabun, 2017-01-16:
NRFgo won't talk to the nRF52 because that's the wrong tool, it's only intended for the nRF24. If you install the nRF5x toolset you should have a fresh set of drivers for working with DK outside of Arduino. There is also an MDK available for Keil if you don't already know that. Both are available on the Nordic site:

[http://www.nordicsemi.com/eng/Products/Bluetooth-low-energy/nRF52832#Downloads](url)

boppana113, 2017-01-16:
I have keil already. Another is issue is that the when I try to hit erase or flash, Keil says there is no J-link device found. I tried to reupload the firmware, but it did not seem to change anything.

dlabun, 2017-01-16:
Which firmware did you try? There's an mbed version and a JLink version.

boppana113, 2017-01-16:
Jlink version.

dlabun, 2017-01-16:
I would recommend trying the nRF5x command line tool and see if you can communicate with the JLink... At least you'll know if it's a communication issue or Keil configuration issue.

dlabun, 2017-01-24:
Closing due to lack of activity

closed 2017-01-24 by dlabun

sandeepmistry, 2017-01-30:
I've added a note to the read me on how to revert to the original driver: https://github.com/sandeepmistry/arduino-nRF5/commit/c5bb63fd4a920fe2e4033a68affdb4a9c428be27

rishi99es, 2017-10-29:
@sandeepmistry 
I tried using Zadig with PCA10040 board. I am trying to revert back to original firmware. I tried your note suggested above. It did not work.
I tried refreshing the Segger J-Link firmware on the board by entering J-Link Bootloader mode, yet I am facing issues in programming the device.

> ERROR: There is no debugger connected to the PC.
> make: xxx [erase] Error 41

This is the nrfjprog error that I get.

pranavb25, 2018-12-22:
I am also facing the same issue. Driver revert back not happening as per README. Has any one been able to find a solution?