sandeepmistry/arduino-nRF5 #82: Where to find examples on how to use BT-LE on the nRF52 chip?

TraxXavier, 2017-01-21:
Where can I find some examples how to get BT-LE to operate on the nRF52?

I would need something as a starting point to be able to read some data out on an android phone

Could anyone please point me to some sample code I could start with?

Thanks
Trax

dlabun, 2017-01-22:
Quoting the readme file:

This Arduino Core does not contain any Arduino style API's for BLE functionality. All the relevant Nordic SoftDevice (S110, S130, S132) header files are included build path when a SoftDevice is selected via the Tools menu.

Recommend BLE Libraries

BLEPeripheral
v0.3.0 and greater, available via the Arduino IDE's library manager.
Supports peripheral mode only.

TraxXavier, 2017-01-22:
Ok this: https://github.com/sandeepmistry/arduino-BLEPeripheral ?
but it does not mentions nRF52 is it compatible with it anyways?

TraxXavier, 2017-01-22:
the lib seams to work fine thanks

closed 2017-01-22 by TraxXavier