sandeepmistry/arduino-nRF5 #83: bluetooth not working

cheeta1, 2017-01-22:
hi sandeep, thanks for the nrf51822 repository. i am new to 51822 chips. i tried blink led and serial codes using  generic nrf51822 board and they work. but when i flash softdevice none of the code seem to work.can you tell me why is that and how can i use the bluetooth functionality using your ble peripheral library ? 

thanks

dlabun, 2017-01-22:
Could provide a little more information? Board you are using, Soft device you selected?

cheeta1, 2017-01-22:
hi dladun,

i am using skylab module skb360, it is based on nrf51822QF AA. i am using softdevice S130. i am using ST link V2 to program it.

dlabun, 2017-01-22:
I don't have one of those modules so it's hard to say what is causing the issue... Just a guess, but it's possible the problem is related to the crystal configuration. I would recommend trying each of the crystal configuration options in the IDE and see if one of them works.

cheeta1, 2017-01-22:
already tried but no luck. i also tried the code on other modules but same problem occur. the modules i have tried so far are given below

http://www.skylab.com.cn/en/productview-69.html
http://www.skylab.com.cn/en/productview-79.html
https://www.aliexpress.com/store/product/1pc-Nordic-BLE-sensor-nRF51822-2-4G-wireless-module-for-Xbee-ble4-0-small-size-SMD/605000_32704439466.html

i am now using wave share module
http://www.waveshare.com/nrf51822-eval-kit.htm

but it show up as cortex -M trying to figure out a way to program them

dlabun, 2017-01-22:
Can you read the entire Nordic part number off the modules? I do not believe the QFAA chips are capable of running some of the soft devices due to a lack of RAM.

cheeta1, 2017-01-22:
they have part number as followings

nrf51822 QFAAH0 1531j4
nrf51822 QFAAH0 1615SD

dlabun, 2017-01-23:
Try the S110 soft device, there was a previous issue opened and they found only S110 worked on the QFAA chips. Unfortunately I think you're running into the same issue due to the lack of RAM.

cheeta1, 2017-01-23:
i tried both S130 and S110 but both doesn't work. there is a guy roger clark i tried his repo with S130 and it worked but it does not have ST link support and i have to use stm controller as black magic probe every time i have to program and i take very long..

rogerclarkmelbourne, 2017-01-23:
S130 does run on the QFAA. 

in SDK 12.1 the S130 SD normally takes 0x1FE8 of RAM for real world have any services.

The most likely reason you can't program is that perhaps the device ID of the devices you are using is not recognised by the BMP.

If the BMP does not recognise the Device ID in the nRF51 it will treat it as a ARM Cortex-M class processor and the method it uses to erase the flash doesnt work.

If you are using the BMP you need to manually run it in GDB and check whether after you scan and attach to the nRF51 that it reports that it has found a nRF51

If not, you need to read the device ID (using GDB) and then update the BMP source code, recompile for this new type of nRF51 and then reflash your BMP.

cheeta1, 2017-01-23:
hi Roger Clark,
i tried S110  with sandeep repo but it does not work. if i use peripherals like serial or i2c or just blink leds it works fine but as soon as i flash softdevice using sandeep repo everything stops working. when i select no softdevice then peripherals start working again.  i tried you repo with S130 and used different BLE example and it works fine

cheeta1, 2017-01-23:
hi Roger Clark,

i just uploaded soft device S110 using black magic probe and the used sandeep repo and it worked. but if i flash S110 using sandeep repo and then use custom code it does not work.

rogerclarkmelbourne, 2017-01-23:
If you device is not recognised you'd need to determine the device ID and then add that device ID to the BMP source and recompile etc

You could try getting the latest source code from Blacksphere's github repo github.com/blacksphere/blackmagicprobe as that may have some more recent nRF51 devices added to the list
(I submitted a PR for a H1 series nRF51 to them a few months ago, but I think you said your device was a H0)

But you will still need to build from source as even though Blacksphere do make precompiled binaries available, they are only updater binaries, and do not contain all the firmware.

So your best option is to recompile it yourself

dlabun, 2017-01-23:
@rogerclarkmelbourne is this a device programming issue or a problem with the core?

rogerclarkmelbourne, 2017-01-23:
@diabun

The issue is the programming hardware as it does intelligent detection of the hardware type ( so that the user does not need to specify device type)

But this fails when a manufacturer brings out a new version of a device and changes the internal Device ID number.

The problem happens on Nordic devices a lot, as Nordic seem to release fixes to their silicon several times a year and hence give these new versions of the nRF5 series a different Device id

So, basically, this is not an issue with the core and from that standpoint should be closed

sandeepmistry, 2017-01-30:
@cheeta1 can you please try with the master version of the core? There was an issue with the flash not being erased before flashing the soft device which has not been released yet.

sandeepmistry, 2017-02-13:
Closing for now due to lack of activity.

closed 2017-02-13 by sandeepmistry